Makuna/Rtc #1: Which RTC.H are you using ?

SoniaZotz, 2015-03-23:
#include <avr/pgmspace.h>
#include <Wire.h>  // must be incuded here so that Arduino library object file references work
#include <RtcDS3231.h>

In file included from DS3231_Alarms.ino:9:0:
C:\Users\Sonia\Documents\Arduino\libraries\Rtc-master/RtcDS3231.h:12:17: fatal error: RTC.h: No such file or directory
 #include <RTC.h>

Makuna, 2015-03-23:
Those were left over lines after a refactor.  I removed them and published the changes.

thanks for report!

Michael

Sent from Windows Mail

From: SoniaZotz
Sent: ‎Sunday‎, ‎March‎ ‎22‎, ‎2015 ‎9‎:‎07‎ ‎PM
To: Makuna/Rtc

#include 
#include // must be incuded here so that Arduino library object file references work
#include 

In file included from DS3231_Alarms.ino:9:0:
C:\Users\Sonia\Documents\Arduino\libraries\Rtc-master/RtcDS3231.h:12:17: fatal error: RTC.h: No such file or directory
#include

SoniaZotz, 2015-03-25:
What I meant was -->>>
In file included from DS3231_Alarms.ino:9:0:
C:\Users\Sonia\Documents\Arduino\libraries\Rtc-master/RtcDS3231.h:12:17: fatal error: RTC.h: No such file or directory

The RTC.h file could not be found

I am also having problems with your Alarm sketch  

#define RtcSquareWavePin 19 // int2  <<<  19 is not an interrupt pin for #define ISC21   5
my understanding is that an interrupt should be generated 
 // Alarm 2 set to trigger at the top of the minute and
// Alarm 1 set to trigger every day 

the output of the SWQ/INT is connected 
Board	int.0	int.1	
Uno,	2	3

Makuna, 2015-03-25:
Please get the latest files for the library.  my library was incorrectly including rtc.h and I have since fixed it.

The sample is valid for the Megao2560 Arduino.

What Arduino do you have? UNO?

What interrupt/pin did you connect the SQW pin?

From: SoniaZotz
Sent: ‎Tuesday‎, ‎March‎ ‎24‎, ‎2015 ‎11‎:‎13‎ ‎PM
To: Makuna/Rtc
Cc: Makuna@hotmail.com

What I meant was -->>>
In file included from DS3231_Alarms.ino:9:0:
C:\Users\Sonia\Documents\Arduino\libraries\Rtc-master/RtcDS3231.h:12:17: fatal error: RTC.h: No such file or directory

The RTC.h file could not be found

I am also having problems with your Alarm sketch 

#define RtcSquareWavePin 19 // int2 <<< 19 is not an interrupt pin for #define ISC21 5
my understanding is that an interrupt should be generated 
// Alarm 2 set to trigger at the top of the minute and
// Alarm 1 set to trigger every day 

the output of the SWQ/INT is connected 
Board int.0 int.1

Uno, 2 3

Makuna, 2015-03-25:
Also note, I just updated the sample to use Arduino calls for interrupts as I was using the lower level ones.  This should make it easier to switch to a Pin and Interrupt for other devices.  I now include a lookup table in the code.

Give it a try.

Sent from Windows Mail

From: Makuna@hotmail.com
Sent: ‎Wednesday‎, ‎March‎ ‎25‎, ‎2015 ‎12‎:‎37‎ ‎AM
To: Makuna/Rtc

Please get the latest files for the library.  my library was incorrectly including rtc.h and I have since fixed it.

The sample is valid for the Megao2560 Arduino.

What Arduino do you have? UNO?

What interrupt/pin did you connect the SQW pin?

From: SoniaZotz
Sent: ‎Tuesday‎, ‎March‎ ‎24‎, ‎2015 ‎11‎:‎13‎ ‎PM
To: Makuna/Rtc
Cc: Makuna@hotmail.com

What I meant was -->>>
In file included from DS3231_Alarms.ino:9:0:
C:\Users\Sonia\Documents\Arduino\libraries\Rtc-master/RtcDS3231.h:12:17: fatal error: RTC.h: No such file or directory

The RTC.h file could not be found

I am also having problems with your Alarm sketch 

#define RtcSquareWavePin 19 // int2 <<< 19 is not an interrupt pin for #define ISC21 5
my understanding is that an interrupt should be generated 
// Alarm 2 set to trigger at the top of the minute and
// Alarm 1 set to trigger every day 

the output of the SWQ/INT is connected 
Board int.0 int.1

Uno, 2 3

SoniaZotz, 2015-03-25:
Thank You Michael,

this is what I tried a while ago

// CONNECTIONS:
// DS3231 SDA --> SDA
// DS3231 SCL --> SCL
// SQW --->  (Pin 3) Don't forget to pullup (4.7k to 10k to VCC)

#include <avr/pgmspace.h>
#include <Wire.h>  // must be incuded here so that Arduino library object
file references work
#include <RtcDS3231.h>

#define ISC10   3  // 0101
#define INT0 0     // UNO external interrupt 0 pin 2
#define INT1 1     // UNO external interrupt 1 pin 3

RtcDS3231 Rtc;

#define RtcSquareWavePin 3 // int0

// marked volatile so interrupt can safely modify them and
// other code can safely read and modify them
volatile uint16_t interuptCount = 0;
volatile bool interuptFlag = false;

// Interupt Service Routine for Int0
ISR(INT0_vect)
{
  // since this interupted any other running code,
  // don't do anything that takes long and especially avoid
  // any communications calls within this routine
  interuptCount++;
  interuptFlag = true;
}

void setup ()
{
  Serial.begin(57600);

  // set the interupt pin to input mode
  pinMode(RtcSquareWavePin, INPUT);

  //--------RTC SETUP ------------
  Rtc.Begin();

  RtcDateTime compiled = RtcDateTime(__DATE__, __TIME__);

  if (!Rtc.IsDateTimeValid())
  {
    Serial.println("RTC lost confidence in the DateTime!");
    Rtc.SetDateTime(compiled);
  }

  RtcDateTime now = Rtc.GetDateTime();
  if (now < compiled)
  {
    Serial.println("RTC is older than compile time!  (Updating DateTime)");
    Rtc.SetDateTime(compiled);
  }

  Rtc.Enable32kHzPin(false);
  Rtc.SetSquareWavePin(DS3231SquareWavePin_ModeAlarmBoth);

  // Alarm 1 set to trigger every day when
  // the hours, minutes, and seconds match
  RtcDateTime alarmTime = now + 88; // into the future
  DS3231AlarmOne alarm1(
    alarmTime.Day(),
    alarmTime.Hour(),
    alarmTime.Minute(),
    alarmTime.Second(),
    DS3231AlarmOneControl_HoursMinutesSecondsMatch);
  Rtc.SetAlarmOne(alarm1);

  // Alarm 2 set to trigger at the top of the minute
  DS3231AlarmTwo alarm2(
    0,
    0,
    0,
    DS3231AlarmTwoControl_OncePerMinute);
  Rtc.SetAlarmTwo(alarm2);

  // throw away any old alarm state before we ran
  Rtc.LatchAlarmsTriggeredFlags();

  // setup external interupt INT 0
  EICRA = 0;              // clear all
  EICRA |= _BV(ISC10);    // falling edge
  EIMSK |= _BV(INT0);     // enable INT0 interrupt
}

void loop ()
{
  if (!Rtc.IsDateTimeValid())
  {
    Serial.println("RTC lost confidence in the DateTime!");
  }

  RtcDateTime now = Rtc.GetDateTime();

  printDateTime(now);
  Serial.println();

  // we only want to show time every 10 seconds
  // but we want to show responce to the interupt firing
  for (int timeCount = 0; timeCount < 20; timeCount++)
  {
    if (Alarmed())
    {
      Serial.print(">>Interupt Count: ");
      Serial.print(interuptCount);
      Serial.println("<<");
    }
    delay(500);
  }
}

bool Alarmed()
{
  bool wasAlarmed = false;
  if (interuptFlag)  // check our flag that gets sets in the interupt
  {
    wasAlarmed = true;
    interuptFlag = false; // reset the flag

    // this gives us which alarms triggered and
    // then allows for others to trigger again
    DS3231AlarmFlag flag = Rtc.LatchAlarmsTriggeredFlags();

    if (flag & DS3231AlarmFlag_Alarm1)
    {
      Serial.println("alarm one triggered");
    }
    if (flag & DS3231AlarmFlag_Alarm2)
    {
      Serial.println("alarm two triggered");
    }
  }
  return wasAlarmed;
}

#define countof(a) (sizeof(a) / sizeof(a[0]))

void printDateTime(const RtcDateTime& dt)
{
  char datestring[20];

  snprintf_P(datestring,
             countof(datestring),
             PSTR("%02u/%02u/%04u %02u:%02u:%02u"),
             dt.Month(),
             dt.Day(),
             dt.Year(),
             dt.Hour(),
             dt.Minute(),
             dt.Second() );
  Serial.print(datestring);
}

Without any success, unfortunately

I'm using the ATmega328P processor and it only has two external interrupts

EICRA – External Interrupt Control Register A
The External Interrupt Control Register A contains control bits for
interrupt sense control.
• Bit 7..4 – Res: Reserved Bits
These bits are unused bits in the ATmega48PA/88PA/168PA/328P, and will
always read as
zero.
• Bit 3, 2 – ISC11, ISC10: Interrupt Sense Control 1 Bit 1 and Bit 0
The External Interrupt 1 is activated by the external pin INT1 if the SREG
I-flag and the corresponding
interrupt mask are set. The level and edges on the external INT1 pin that
activate the
interrupt are defined in Table 12-1. The value on the INT1 pin is sampled
before detecting
edges. If edge or toggle interrupt is selected, pulses that last longer
than one clock period will
generate an interrupt. Shorter pulses are not guaranteed to generate an
interrupt. If low level
interrupt is selected, the low level must be held until the completion of
the currently executing
instruction to generate an interrupt.
• Bit 1, 0 – ISC01, ISC00: Interrupt Sense Control 0 Bit 1 and Bit 0
The External Interrupt 0 is activated by the external pin INT0 if the SREG
I-flag and the corresponding
interrupt mask are set. The level and edges on the external INT0 pin that
activate the
interrupt are defined in Table 12-2. The value on the INT0 pin is sampled
before detecting
edges. If edge or toggle interrupt is selected, pulses that last longer
than one clock period will
generate an interrupt. Shorter pulses are not guaranteed to generate an
interrupt. If low level
interrupt is selected, the low level must be held until the completion of
the currently executing
instruction to generate an interrupt.
Bit 7 6 5 4 3 2 1 0
(0x69) – – – – ISC11 ISC10 ISC01 ISC00 EICRA
Read/Write R R R R R/W R/W R/W R/W
Initial Value 0 0 0 0 0 0 0 0

let me look at you new sample  and see it I can make sense of it

Thanks again

Sonia

SoniaZotz, 2015-03-25:
Michael,

Thank you for your help

with this

//#define RtcSquareWavePin 19 // Mega2560
//#define RtcSquareWaveInterrupt 4 // Mega2560

#define RtcSquareWavePin 2 // Output pin from DS3231 Pulled up on Board 4.7
K --->>>  to UNO ATMeg328p pin 2
#define RtcSquareWaveInterrupt 0 // UNO ATMeg328p interupt service 0 or 1
for UNO set 0

it now seems to work

03/25/2015 16:40:45
03/25/2015 16:40:55
alarm two triggered
>>Interrupt Count: 1<<
03/25/2015 16:41:05
03/25/2015 16:41:15
03/25/2015 16:41:25
03/25/2015 16:41:35
03/25/2015 16:41:45
03/25/2015 16:41:55
alarm two triggered
>>Interrupt Count: 2<<
03/25/2015 16:42:05
alarm one triggered
>>Interrupt Count: 3<<
03/25/2015 16:42:15
03/25/2015 16:42:25
03/25/2015 16:42:35
03/25/2015 16:42:45
03/25/2015 16:42:55
alarm two triggered
>>Interrupt Count: 4<<
03/25/2015 16:43:05
03/25/2015 16:43:15
03/25/2015 16:43:25

SoniaZotz, 2015-03-26:
How does one set alarm times  other that what is in the example?

ie ....AlarmOne  1:45:21  Saturday
and alarmTwo  13 hr  34 min 42 sec  on Wednesday  Aug 21 2016

Makuna, 2015-03-27:
1:45:21 Saturday would be
 
 DS3231AlarmOne alarm1(6, // day of week, Sunday is 0, Monday is 1, Tuesday 2 ...
            1
            45,
            21,            DS3231AlarmOneControl_HoursMinutesSecondsDayOfWeekMatch);
 
13 hr  34 min 42 sec  on Wednesday  Aug 21 2016
(NOTE: it doesn't support years or months for alarms, just day of week or day of the month, further alarm two doesn't support seconds, only minutes)
    DS3231AlarmTwo alarm2(21
            13,
            34,
DS3231AlarmTwoControl_HoursMinutesDayOfMonthMatch

 
 
 
 
 
 
 
 
 
 

 
Date: Thu, 26 Mar 2015 08:48:32 -0700

closed 2015-06-15 by Makuna