Makuna/Rtc #3: RtcDateTime.Year return 2115 instead 2015.

far5893, 2015-07-28:
String message = "<?xml version=\"1.0\" encoding=\"UTF-8\" standalone=\"yes\"?>\n\r<time>\n\r";
       message += "<sec>" + String(now.Second()) + "</sec>\n";
       message += "<min>" + String(now.Minute()) + "</min>\n";
       message += "<hour>" + String(now.Hour()) + "</hour>\n";
       message += "<day>" + String(now.Day()) + "</day>\n";
       message += "<month>" + String(now.Month()) + "</month>\n";
       message += "<year>" + String(now.Year()) + "</year>\n";
       message += "<timestamp>" + String(now.TotalSeconds())+ "</timestamp>\n";
       message += "</time> \n\r";

return 
....
<year>2115</year>
.....

Makuna, 2015-08-01:
Can you give me more information.
How was this RtcDateTime retrieved?  "now" is a RtcDateTime correct? 
From which model of rtc device?  
How was the date and time set on the device?

far5893, 2016-01-13:
I set ds3231  under linux (cubieboard), i suppose that is necessary to reset century flag but i don't find how to reset ,datasheet does not be clear.

Makuna, 2016-01-13:
If you examine the code for the ds3231.cpp (RtcDS3231::GetDateTime) you will see that its base year is 2000, and that it will only add 100 if the century bit is set, thus giving a possible range of 2000 - 2199. 

What did you change to support their code?

In my testing I am not seeing what you are seeing, but then I am not using Linux to set it and I don't know what their code is doing; but I can't see how anything they could do would allow my code to come up with a range outside this.

The only thing I can think of is that they are setting the bit when they shouldn't.  Could you get a raw copy of the bytes from a date dump and include those?

Just add a method to like...

```
Wire.beginTransmission(DS3231_ADDRESS); 
Wire.Write(DS3231_REG_TIMEDATE);            
Wire.endTransmission(); 
 
Wire.requestFrom(DS3231_ADDRESS, 7); 
for (int i = 0; i < 7; i++)
{
  uint8_t rawbyte = Wirte.Read();
  Serial.Write(rawbyte, HEX); 
}
```

far5893, 2016-01-14:
Serial.println("-----Start I2C Debug --------");
        Wire.beginTransmission(DS3231_ADDRESS);
        Wire.write(DS3231_REG_TIMEDATE);
        Wire.endTransmission();

        Wire.requestFrom(DS3231_ADDRESS, 7);
        for (int i = 0; i < 7; i++)
        {
          uint8_t  rawbyte ;
          rawbyte = Wire.read();

          Serial.print(String(rawbyte)+" ");
        }
        Serial.println("\n-----Stop I2C Debug --------");

Output(decimal) : 
-----Start I2C Debug --------
38 73 8  5 20 129 22 
-----Stop I2C Debug --------

in ds1307.c (kernel 3.4.79)
static int ds1307_set_time(struct device *dev, struct rtc_time *t)
{
        struct ds1307   *ds1307 = dev_get_drvdata(dev);
        int             result;
        int             tmp;
        u8              *buf = ds1307->regs;

        dev_dbg(dev, "%s secs=%d, mins=%d, "
                "hours=%d, mday=%d, mon=%d, year=%d, wday=%d\n",
                "write", t->tm_sec, t->tm_min,
                t->tm_hour, t->tm_mday,
                t->tm_mon, t->tm_year, t->tm_wday);

        buf[DS1307_REG_SECS] = bin2bcd(t->tm_sec);
        buf[DS1307_REG_MIN] = bin2bcd(t->tm_min);
        buf[DS1307_REG_HOUR] = bin2bcd(t->tm_hour);
        buf[DS1307_REG_WDAY] = bin2bcd(t->tm_wday + 1);
        buf[DS1307_REG_MDAY] = bin2bcd(t->tm_mday);
        buf[DS1307_REG_MONTH] = bin2bcd(t->tm_mon + 1);

        /* assume 20YY not 19YY */
        tmp = t->tm_year - 100;
        buf[DS1307_REG_YEAR] = bin2bcd(tmp);

        switch (ds1307->type) {
        case ds_1337:
        case ds_1339:
        case ds_3231:
                buf[DS1307_REG_MONTH] |= DS1337_BIT_CENTURY;
                break;
        case ds_1340:
                buf[DS1307_REG_HOUR] |= DS1340_BIT_CENTURY_EN
                                | DS1340_BIT_CENTURY;
                break;
        case mcp7941x:
                /*
                 * these bits were cleared when preparing the date/time
                 * values and need to be set again before writing the
                 * buffer out to the device.
                 */
                buf[DS1307_REG_SECS] |= MCP7941X_BIT_ST;
                buf[DS1307_REG_WDAY] |= MCP7941X_BIT_VBATEN;
                break;
        default:
                break;
        }

        dev_dbg(dev, "%s: %02x %02x %02x %02x %02x %02x %02x\n",
                "write", buf[0], buf[1], buf[2], buf[3],
                buf[4], buf[5], buf[6]);

        result = ds1307->write_block_data(ds1307->client,
                ds1307->offset, 7, buf);
        if (result < 0) {
                dev_err(dev, "%s error %d\n", "write", result);
                return result;
        }
        return 0;
}
It set bit century to 0x80 (or-ing)

Makuna, 2016-01-24:
If you look at the raw data, the century bit is set so the library is returning the correct date.

The code above that you use to set the date is flawed.  It is setting the century bit always.  It uses or operator to set the bit, this is how you do it c/c++.

closed 2016-01-24 by Makuna