Makuna/Rtc #5: Esp8266 Support

jorgegarciadev, 2016-01-24:
I've been trying your library but i'm not able to read time from my DS1307. Could you share the module or the circuit that you have used for the tests?

Makuna, 2016-01-24:
It is pretty simple to connect.  I have just updated the readme and the sample files to include a little more details.  But the connections were at the top of the samples.

Just connect the SDA and SCL from the RTC to the pins that match them.  For the esp8266, the defaults are SDA == GPIO04 and SCL == GPIO05.  Which pins varies on which version of the Esp8266 you are using.  For the ESP01, you will need to ignore the defaults and the samples include a line of code you can uncomment.  NOTE:  I updated the samples to not override these for all ESP8266 like it used it.

Also, for the DS1307, you need to power it with 5v, the datasheet doesn't state that it is 3.3v tolerant for power, but the IO pins seem to be tolerant of 3.3v so I don't think you will need to do any level shifting for them.  Make sure the GND on the RTC is also connected to the GND on Esp just in case you are powering the RTC from another power source.

jorgegarciadev, 2016-01-24:
Thank you for your answer. That's how I've connected the DS1307 to my NodeMCUv2. In different tests I have added 2k2 pullup resistors to the i2c lines, even used a level shifter without luck.

The library works with the same DS1307 and an Arduino Nano, also with the NodeMCUv2 and a DS3231 module. I don't know what I'm missing. I'm having the same problem with every RTC library out there and the DS1307.

I'm not using a RTC module, built a circuit following [this schematic](https://github.com/adafruit/DS1307-breakout-board). If you're using and RTC module could you share the model?

Makuna, 2016-01-25:
I use the NodeMcuV2 board and even tested it this morning.

Note, the actual pins used can be almost any of them that aren't already used.  By default they aren't the i2c lines, the are just the GPIO4 and GPIO5.

I will see if I can dig up where I bought it, its been over 4 years since I got it.

Makuna, 2016-01-25:
http://www.dx.com/p/tiny-rtc-real-time-clock-module-for-arduino-red-161172#.VqWQEE6IZDs
This is pretty close to what I have.  Mine has a few more unpopulated locations for components, but the similarity is 99% inclusive the labels.

jorgegarciadev, 2016-01-25:
Thank you. I found the [schematic](http://www.dipmicro.com/store/DS1307BAT-MOD) for that module and will try it later. I also asked some friends if they have it. Will report back when I make some tests and get some answers.

jorgegarciadev, 2016-03-02:
Sorry, It took some more time than I thought. I have done some tests with this [i2c scanner](http://www.14core.com/scanning-i2c-device-address-for-arduino/), the esp8266 can find the ds1307 at address 0x68 using 2.2k pullup resistors, It can even set the time but the ds1307 doesn't increase the count, I don't know whats going on.

The TinyRTC uses a ds1307z, don't know how different it is from the ds1307 but it works with the library and esp8266. Since the ds1307z it's too small for my soldering skills I'm moving to the ds1302, I made it work this afternoon with the NodeMCUv2 without problems.

Thanks for your time and your help. I'm closing the issue.

closed 2016-03-02 by jorgegarciadev