Makuna/Rtc #9: How to set rtc with NTP?

spirkaa, 2016-01-26:
Can you explain how to set rtc1307 using your lib and **epoch** from [NTPClient esp8266wifi example](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/NTPClient/NTPClient.ino#L105), please?

`Rtc.SetDateTime(epoch); ` results year 2046. How to correct this?

Makuna, 2016-01-26:
First, if you look at the method Rtc::SetDateTime, you will see that it takes a RtcDateTime object, not an unsigned long.
If you then look at the RtcDateTime object, you will see that the constructor takes a uint32_t which is what is constructed for you when you try to pass the unsigned long to Rtc::SetDateTime.  This argument is the seconds from year 2000, not "Epic" time.
Look further into RtcDateTime, you will see the method InitWIthEpic32Time() or InitWithEpic64Time(), these are what you want to use to init with Epoc time.
So, init an RtcDateTime by calling InitWIthEpic32Time() or InitWithEpic64Time(), then pass that object into the Rtc::SetDateTime.

closed 2016-01-26 by Makuna

NarinLab, 2020-11-20:
Here is a function script to update RTC from NTP, change GMT_OFFSET_SEC and DAYLIGHT_OFFSET_SEC to suit your location.
```
void system_ntp_rtc(){
  configTime(GMT_OFFSET_SEC, DAYLIGHT_OFFSET_SEC, "pool.ntp.org");
  struct tm timeinfo;
  if(getLocalTime(&timeinfo)){
    RtcDateTime ntpdt = RtcDateTime(timeinfo.tm_year + 1900, timeinfo.tm_mon + 1, timeinfo.tm_mday, timeinfo.tm_hour, timeinfo.tm_min, timeinfo.tm_sec);
    RTC.SetDateTime(ntpdt);
  }
}
```

jumblies, 2021-03-14:
If you were stymied by reading this initially,  this code should set the RTC on a valid NTP packet using the NTPClient library.  
```
    timeClient.begin();
    timeClient.update();
    RtcDateTime utc;
    utc.InitWithEpoch32Time(timeClient.getEpochTime());
    Rtc.SetDateTime(utc);
```