Makuna/Rtc #16: Issue in AsFloat() method

anteph, 2016-05-27:
Hi,

I've detected an issue when calling the AsFloat() method to get the temperature from a RtcTemperature object.

The sum of the decimal and integer parts are being performed as if they are two integers. For example, if the integer part is 22 and the decimal 0.25, the output of the AsFloat() is 47.00.

If I call the AsWholeDegrees() and GetFractional() methods and perform the sum myself, it works fine. 

I'm using a NodeMCU v2 for testing.

Best regards,
Nuno Santos
Blog: https://techtutorialsx.wordpress.com/

Makuna, 2016-05-27:
Could you confirm what version of the library that you have?

1) open Arduino/Libraries/Rtc/libraries.properties file with a text editor and it will have a field for version.
2) Check the Arduino IDE library manager and what version does it show?

anteph, 2016-05-28:
In the libraries.properties file it has version 1.0 and on library manager it also has version 1.0.0. At the time of installation, using the IDE library manager, it was the only version available.

Makuna, 2016-05-28:
This is your problem.  The current version is 1.0.1; which includes a fix for the problem you are having.

Normally the IDE will notify you of an update.  But you can force it by going to the library manager and filter the list by "installed" and selecting update button.

closed 2016-05-28 by Makuna

anteph, 2016-05-29:
It solved the problem, it's working perfectly. Thanks for your help.

Best regards,
Nuno Santos
Blog: https://techtutorialsx.wordpress.com/