Makuna/Rtc #26: Not a real issue (I hope): I forked and modified your library

smz, 2017-02-28:
Hello @Makuna, I just want you to know  that I just uploaded in my GitHub an Arduino library that essentially is a fork of your work (see: https://github.com/smz/Arduino-RTCtime).

My need was using the classic time.h structures/methods instead of your RtcDateTime class, so I did the relevant changes, found it working good and decided to contribute back to the community with my version.

I hope this is not a problem for you.

Please note that in my README.md I recognized the origin of my work and also suggested donating to **your** project.

Regards,

Sergio Manzi

Makuna, 2017-02-28:
The only real issue is consumer confusion and lag in fixes between the two.  But I am sure we can work on it.

Have you thought about just merging your changes back into my original?  Not sure its right thing, just wondering if you thought about it and what you learned.

smz, 2017-02-28:
Hello Michael,

Yes I did consider that, but I think that this potentially will create **even more** confusion as there will be different methods for handling the same things using different data types.

Beside that the library will be more "bloated" and heavy on the scarce Arduino resources.

I will appreciate if you would look at my version and if you find it can really be merged with yours, I will be happy about that and will close my project, no problem at all.

Cheers,

Sergio

keyeh, 2017-02-28:
+1 to using the standard `Time.h` library. It allows the use of other libaries such as [NtbClientLib](https://github.com/gmag11/NtpClient) easily without having to convert the datastructure back and forth.

Makuna, 2017-02-28:
You both mention standard time library.  Arduino doesn't ship with a time library built in, so its not really a standard.  What library are you talking about?

I found this one [GitHub/Feilipu/Arduino RTC Library](https://github.com/feilipu/Arduino_RTC_Library) but I am not sure I would call it standard, as the Arduino IDE even tags it with incompatible and it really doesn't follow standard Arduino style at all.

smz, 2017-02-28:
It is the usual `Time.h` library found on all *nix OS.

It comes with the Arduino IDE too as `\Arduino\hardware\tools\avr\avr\include\time.h`

smz, 2017-02-28:
... more information can be found in Wikipedia too (https://en.wikipedia.org/wiki/C_date_and_time_functions), and yes, it is definitely part of a standard:  `ISO/IEC 9899:1999 specification`

Makuna, 2017-02-28:
please understand, standard within the context of this "library" is Arduino, not *nix.  Arduino is by no means a *nix system. Time.h maybe a Unix standard; so please be specific when using the word "standard".

After reviewing that API, I am not so interested anymore.  Its a very UGLY API and Arduino is aimed at less advanced programmers and I don't want this library to become unusable by the novices; which I believe will be the case if I use that time API standard.

smz, 2017-03-01:
... and to be fair, it must be said that the AVR implementations differ from the standard in its epoch: the standard epoch for time_t is January 1, 1970, while the AVR standard for time_t is January 1, 2000

smz, 2017-03-01:
Michael, I was wrong with my assertion (and I edited it...): it is **not* a *nix standard, but a **C** standard...

Makuna, 2017-03-01:
it still doesn't it make it right (for novices).  That was and is the goal of Arduino is to make C++ more approachable to novices.  And I have tried hard to maintain that.  The old C standards of strange acronyms and spelling of words is so 1990s.

I think I will add a link somewhere here to mention if they want support for the C standard time; I will point them to your library.

I will use this issue to track that.

smz, 2017-03-01:
I have no problem with that, actually I appreciate it, but I think that's a mistake from your part (_and please do not take an offence from my assertion, I really value your work here, sincerely_) diverging from what's considered a consolidated programming standard...

Makuna, 2017-03-01:
BTW, things to know...
* from time.h, time_t is not Epoch time.   
* from my lib, RtcDateTime can be initialized with a time_t directly, as it is seconds from 2000, same as what RtcDateTime takes.  
* if you need a time_t, just call RtcDateTime method TotalSeconds().  

so using my library with time.h really isn't that hard or complex, its just not documented well.

so with a rtc object,
```
time_t time = rtc.GetDateTime().TotalSeconds(); // retrieve date and time as time_t

rtc.SetDateTime(RtcDateTime(time)); // set date and time with a time_t
```

smz, 2017-03-01:
Hello!

Yes, correct, Arduino's time.h doesn't use the Unix epoch, but its own epoch defined as 2000-01-01T00:00:00.00Z  (_the Y2K epoch_). There are other implementations around that do the same, like AppleSingle, AppleDouble, PostgreSQL, ZigBee UTCTime

See: https://en.wikipedia.org/wiki/Epoch_(reference_date)

You can convert to/from the Unix epoch by adding/subtracting the predefined UNIX_OFFSET constant or to/from the NTP epoch by adding/subtracting NTP_OFFSET.

As for the rest, of course you can perform whatever chronological computation using whatever library/reference you wish: it's only a matter of taste. Personally I prefer to stick to the standard, so as to make my programs/libraries a little bit more portable (_think of porting a program/library from Arduino to Raspberry or vice-versa..._)

Makuna, 2017-03-04:
I updated the Wiki to discuss this topic [here](https://github.com/Makuna/Rtc/wiki/FAQ-%234)

smz, 2017-03-04:
Hello, Michael!

I'm totally fine with what you wrote in the Wiki. To be honest I didn't even expected a mention to my fork and I really appreciate it.

Of course if there is any bit of what I did that you would like to incorporate in your library, please feel absolutely free to do that without any further mention to my fork.

Just a couple of minor caveat:
 - My library is not listed yet in the Library Manager, I can't guarantee it will ever be, and to be totally honest I've yet to see what the process to be listed is! (...but I'll try...)
- Link to my repo is https://github.com/smz/Arduino-RTCtime (the difference being that "**s**mz", has a small "s" and not a capital "S". It doesn't seems to matter to GitHub, but... just in case!

Thank-you very much,

Sergio

P.S.: Feel free to close this "_non-issue_" if you think there isn't anything more to discuss about this matter here. Have a nice time!

smz, 2017-03-04:
P.S.: what I'm thinking to do next (_and I'm not totally sure it's a good idea..._) is to **not** using different includes depending on which RTC hardware you have, but just a single include and to apply the correct code based on either of the two:
- a "define" for choosing the correct hardware
- a parameter (from an enum type) to the object instantiation method

e.g:
~~~
#include <RTCtime.h>
#define RTC_HARDWARE DS3231
MyRtc = RTCtime<TwoWire> Rtc(Wire);
~~~

or:

~~~
#include <RTCtime.h>
MyRtc = RTCtime<TwoWire> Rtc(Wire, DS3231);
~~~

I think that in theory (and at first sight) the first solution would have the benefit of minimizing the generated code, while the second would permit to have more than one different RTC using different hardware, and I'm fancying about introducing a "soft RTC", that will need to sync with the "real" RTC (or NTP, or GPS) just at startup (and every "x" time, to not drift too much). Of course NTP and GPS could be defined as different "hardware" supporting the class.

Makuna, 2017-03-05:
The better model, one of which I am been contemplating and not found the time yet to implement, is to provide the device as a Feature template object.

```
RtcTime<TwoWire, RtcDs3231Feature> Rtc(Wire);
```

The code will compile down smaller.

smz, 2017-03-05:
OK, I think it's time for me to confess: my experience in OOP is quite limited.

I've used OOP techniques only with PHP, in the context of web applications, and when it comes to the "C" ecosystem, well... think that when I have doubts I reach for my old K&R, first edition, in my library!

So, TBH, I've difficulties understanding what you said above, although *I think* I have a *very rough* idea of what you meant.

Nevertheless I'm much interested in expanding my knowledge in the field, so... I'll study!

Have a nice Sunday!

Makuna, 2017-03-05:
A must read book for all C++ developers: "Modern C++ Design". ISBN-13: 978-0201704310; ISBN-10: 0201704315.

smz, 2017-03-05:
Thanks for the tip, I will surely do!

smz, 2017-03-05:
it's this one https://www.amazon.it/dp/0201704315, isn't it? What's your opinion: is it readable on the Kindle (*quite cheaper*) or is it better sticking to the Gutenberg way?

Makuna, 2017-03-05:
That's it.  If you like digital, go digital.

smz, 2017-03-05:
Actually I'd rather have it printed (and have it sitting side by side to the good ol' k&r!), but I like paper in my wallet too, so... I don't know... I'll thinks it over... :-)
tnx agn.

shaddow1201, 2017-03-08:
What is the base reference point for DayOfWeek(), and/or can this be specified? (ie. which day of the week is day 1?)   So far I've found the library simple to use, appreciate!

Makuna, 2017-03-08:
In my library, day of week value of zero is Sunday.

closed 2017-03-08 by Makuna