Makuna/Rtc #27: Use hour minute in variable

maxholgasson, 2017-03-24:
Hi,

how can I use the time and store hour and minute in variables?
In the DS3231 Simple - Example is only a Serial print

Thanks

Makuna, 2017-03-24:
This is less an issue and more a simple question, better suited for the gitter channel.
[image]

Just use the access members for hour and minute from the RtcDateTime object like the example does to print it, but just assign it to a variable...
```
RtcDateTime now = Rtc.GetDateTime();
uint8_t hour = now.Hour();
uint8_t minute = now.Minute();
```

maxholgasson, 2017-03-24:
Thanks a lot, I will try this.  What is the gitter channel?

Makuna, 2017-03-24:
Follow the link I included (the button).  Gitter is a chat system tied to github projects.  Great for questions and simple issues to reduce "noise" in the GitHub issues area.

closed 2017-03-31 by Makuna