Makuna/Rtc #29: Errores compilacion

agcII, 2017-03-31:
Arduino:1.8.0 (Windows 10), Tarjeta:"Arduino Due (Programming Port)"

In file included from C:\Users\JuanJo\Documents\Arduino\DS3231_Simple\DS3231_Simple.ino\DS3231_Simple.ino.ino:24:0:

C:\Users\JuanJo\Documents\Arduino\libraries\Rtc_by_Makuna\src/RtcDS3231.h:43:48: error: '_BV' was not declared in this scope

 const uint8_t DS3231_AIEMASK = (_BV(DS3231_A1IE) | _BV(DS3231_A2IE));

                                                ^

C:\Users\JuanJo\Documents\Arduino\libraries\Rtc_by_Makuna\src/RtcDS3231.h:43:67: error: '_BV' was not declared in this scope

 const uint8_t DS3231_AIEMASK = (_BV(DS3231_A1IE) | _BV(DS3231_A2IE));

                                                                   ^

C:\Users\JuanJo\Documents\Arduino\libraries\Rtc_by_Makuna\src/RtcDS3231.h:44:46: error: '_BV' was not declared in this scope

 const uint8_t DS3231_RSMASK = (_BV(DS3231_RS1) | _BV(DS3231_RS2));

                                              ^

C:\Users\JuanJo\Documents\Arduino\libraries\Rtc_by_Makuna\src/RtcDS3231.h:44:64: error: '_BV' was not declared in this scope

 const uint8_t DS3231_RSMASK = (_BV(DS3231_RS1) | _BV(DS3231_RS2));

                                                                ^

C:\Users\JuanJo\Documents\Arduino\libraries\Rtc_by_Makuna\src/RtcDS3231.h:52:47: error: '_BV' was not declared in this scope

 const uint8_t DS3231_AIFMASK = (_BV(DS3231_A1F) | _BV(DS3231_A2F));

                                               ^

C:\Users\JuanJo\Documents\Arduino\libraries\Rtc_by_Makuna\src/RtcDS3231.h:52:65: error: '_BV' was not declared in this scope

 const uint8_t DS3231_AIFMASK = (_BV(DS3231_A1F) | _BV(DS3231_A2F));

                                                                 ^

C:\Users\JuanJo\Documents\Arduino\libraries\Rtc_by_Makuna\src/RtcDS3231.h: In member function 'bool RtcDS3231<T_WIRE_METHOD>::IsDateTimeValid()':

C:\Users\JuanJo\Documents\Arduino\libraries\Rtc_by_Makuna\src/RtcDS3231.h:239:41: error: there are no arguments to '_BV' that depend on a template parameter, so a declaration of '_BV' must be available [-fpermissive]

         return !(status & _BV(DS3231_OSF));

                                         ^

C:\Users\JuanJo\Documents\Arduino\libraries\Rtc_by_Makuna\src/RtcDS3231.h:239:41: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)

C:\Users\JuanJo\Documents\Arduino\libraries\Rtc_by_Makuna\src/RtcDS3231.h: In member function 'bool RtcDS3231<T_WIRE_METHOD>::GetIsRunning()':

C:\Users\JuanJo\Documents\Arduino\libraries\Rtc_by_Makuna\src/RtcDS3231.h:245:40: error: there are no arguments to '_BV' that depend on a template parameter, so a declaration of '_BV' must be available [-fpermissive]

         return !(creg & _BV(DS3231_EOSC));

                                        ^

C:\Users\JuanJo\Documents\Arduino\libraries\Rtc_by_Makuna\src/RtcDS3231.h: In member function 'void RtcDS3231<T_WIRE_METHOD>::SetIsRunning(bool)':

C:\Users\JuanJo\Documents\Arduino\libraries\Rtc_by_Makuna\src/RtcDS3231.h:252:37: error: there are no arguments to '_BV' that depend on a template parameter, so a declaration of '_BV' must be available [-fpermissive]

             creg &= ~_BV(DS3231_EOSC);

                                     ^

C:\Users\JuanJo\Documents\Arduino\libraries\Rtc_by_Makuna\src/RtcDS3231.h:256:36: error: there are no arguments to '_BV' that depend on a template parameter, so a declaration of '_BV' must be available [-fpermissive]

             creg |= _BV(DS3231_EOSC);

[... 199 lines cut ...]

DS3231_Simple.ino:99: error: expected primary-expression before '.' token

     RtcDateTime now = Rtc.GetDateTime();

                          ^

DS3231_Simple.ino:103: error: expected primary-expression before '.' token

     RtcTemperature temp = Rtc.GetTemperature();

                              ^

Se encontraron múltiples librerías para "RtcDS3231.h"
Usado: C:\Users\JuanJo\Documents\Arduino\libraries\Rtc_by_Makuna
 No usado: C:\Users\JuanJo\Documents\Arduino\libraries\RTCtime
 No usado: C:\Users\JuanJo\Documents\Arduino\libraries\RTCtime
 No usado: C:\Users\JuanJo\Documents\Arduino\libraries\RTCtime
 No usado: C:\Users\JuanJo\Documents\Arduino\libraries\RTCtime
exit status 1
'RtcDS3231<TwoWire> Rtc' redeclared as different kind of symbol

Este reporte podría tener más información con
"Mostrar salida detallada durante la compilación"
opción habilitada en Archivo -> Preferencias.

Makuna, 2017-03-31:
It seems the _BV (bit vector) macro is not defined for that board anymore.  
In your primary sketch, before including the rtc header define the following...
```
#define _BV(b) (1UL << (b))
#include <RtcDs3231.h> // in your sketch, before this include, include the above
```

agcII, 2017-03-31:
Yes, it works.
and I need to change.
RtcDS3231<TwoWire> Rtc(Wire);
to
RtcDS3231<TwoWire> Rtc2(Wire);

for redefinition errors.

agcII, 2017-03-31:
Thanks.

2017-03-31 19:15 GMT+02:00 Juan Jose Cabrero <agcfanatic@gmail.com>:

Makuna, 2017-03-31:
This will work around the issue for now, I will leave this issue open to track making a permanent fix.

Makuna, 2017-03-31:
2.0.1 includes the fix

closed 2017-03-31 by Makuna