Makuna/Rtc #32: Using memory at RTC3231

jcmojj, 2017-04-06:
Makuna,

I am using nodemcu 1.0 with RTC3231 that has much greater precision than RTC1307.
I spent all the day trying to find out why my nodemcu 1.0 eeprom start to have problem to save some bytes. I just find out that the 10000 cycles it is supposed to hold on can be hold all together. I tried to program in a way that I save only one byte at time to save cycles, but it doesn't matter because it save it all together every time you do EEPROM.commit() or EEPROM.end(). I am telling you that because I just find out that the RTC3231 I am using with your lib RTC have one memory chip called AT24C32 that is a EEPROM with more than 1 million cycles.
I found that description for that chip:

> The AT24C32/64 provides 32,768/65,536 bits of serial electrically erasable and programmable
> read only memory (EEPROM) organized as 4096/8192 words of 8 bits
> each. The device’s cascadable feature allows up to 8 devices to share a common 2-
> wire bus. The device is optimized for use in many industrial and commercial applications
> where low power and low voltage operation are essential. The AT24C32/64 is
> available in space saving 8-pin JEDEC PDIP, 8-pin JEDEC SOIC, 8-pin EIAJ SOIC,
> and 8-pin TSSOP (AT24C64) packages and is accessed via a 2-wire serial interface.
> In addition, the entire family is available in 2.7V (2.7V to 5.5V) and 1.8V (1.8V to 5.5V)
> versions.

I check your lib and I didn't find anything regarding the memory at your lib. Do you have it? If not, do you plan to do it? It would be great to have it since you did a great job with your lib that works very well with the esp8266. I checked your lib info when you look for lib at arduino IDE and you say: **Includes deep support of modules features, including temperature, alarms and memory storage if present. Tested on esp8266.** I think its the only thing missing!

That library can save at memory, if you need some reference it may help [https://github.com/Naguissa/uRTCLib](url)
Anyway, thank you for the collaboration

Makuna, 2017-04-06:
The actual Rtc3231 chip doesn't have memory support on it that is exposed.  It just happens that the module you have has both the rtc3231 and the AT24cXX on it.

What you need is another library that supports using the at24cXX.

I don't actually have one of these myself.

closed 2017-04-06 by jcmojj

Makuna, 2018-09-16:
Just wanted to add an update to this topic.  The library now includes support for the at24cxx.  It is not exposed from the rtc3241 object, but a separate object is available to access the eeprom memory.