Makuna/Rtc #37: GetIsRunning() returns 0 after SetDateTime from NTP event

cholokla, 2017-06-08:
Hi,

I'm periodically calling `RTCDS3231.SetDateTime(RtcDateTime(year(), month(), day(), hour(), minute(), second())); ` in NTP time sync event handler to update RTC time from internet.
The problem is after such operation `RTCDS3231.GetIsRunning()` and 'RTCDS3231.IsDateTimeValid()' functions are starting to return FALSE status.
I've even tried to call ` RTCDS3231.SetIsRunning(true)` after that but without any result.
RTC seems to initialize succesfully and returns right time just after boot.
What can be wrong?

Greetings,
robert

ESP8266 ESP-12F / Arduino IDE 1.6.12

Makuna, 2017-06-08:
In the NTP sync event handler might be the problem.  In most callbacks from the network, you need to avoid calling anything of any length (like serial or wire classes).  Try storing the time and a flag that states you need to set the date/time, and then in your loop check for this flag, set date time, and clear the flag.

Also, Is the battery in good condition?  A battery that is on the edge causes problems also.

cholokla, 2017-06-08:
Thanks for instant response :)
Battery is new and time sync is done in main loop just after status flag from event handler is set.
One thing is I'm using Serial.print in NTP event handler so I'll try to remove it and see if it corrects the problem.

cholokla, 2017-06-11:
I've removed all Serial calls from NTP event handler but problem still persists (though I have to wait few or more hours).

After GetIsRunning() and IsDateTimeValid() functions start to consequently return false statuses I restart the controller and just after boot function 'printDateTime()' taken from the example prints '85/165/2009 37:165:':

> RTC Time invalid, setting compilation time.
RTC was not running, starting now
RTC time: 85/165/2009 37:165:

Part of my code is attached.
After sync event occured I wait one minute to be sure time is correct.

[RTC.txt](https://github.com/Makuna/Rtc/files/1066290/RTC.txt)
[NTP.txt](https://github.com/Makuna/Rtc/files/1066293/NTP.txt)

Makuna, 2017-06-11:
What is the debug output of your date time from NTP before you call setdatetime?

Setting an incorrect date and time will also cause this.

cholokla, 2017-06-12:
With NTP DEBUG flag enabled.
Sometimes 'GetIsRunning Before' shows 0 just at the beginning, maybe it's NTP client library error.

```
    Serial.print("Syncing RTC time: ");
    Serial.print(day());
    Serial.print("/");
    Serial.print(month());
    Serial.print("/");
    Serial.print(year());
    Serial.print(" ");
    Serial.print(hour());
    Serial.print(":");
    Serial.print(minute());
    Serial.print(":");
    Serial.println(second());

    Serial.print("GetIsRunning Before: ");
    Serial.println(RTCDS3231.GetIsRunning());
    Serial.print("IsDateTimeValid Before: ");
    Serial.println(RTCDS3231.IsDateTimeValid());

    RTCDS3231.SetDateTime(RtcDateTime(year(), month(), day(), hour(), minute(), second()));

    Serial.print("GetIsRunning After: ");
    Serial.println(RTCDS3231.GetIsRunning());
    Serial.print("IsDateTimeValid After: ");
    Serial.println(RTCDS3231.IsDateTimeValid());

    if (!RTCDS3231.GetIsRunning()) {
        RTCDS3231.SetIsRunning(true);
    }

    rtcTimeValid = RTCDS3231.IsDateTimeValid();
    syncNtpRtc = false;

    Serial.print("GetIsRunning End: ");
    Serial.println(RTCDS3231.GetIsRunning());
    Serial.print("IsDateTimeValid End: ");
    Serial.println(RTCDS3231.IsDateTimeValid());
```

> First sync! 21:26:09 12/06/2017
Succeccful NTP sync at 21:26:09 12/06/2017
Syncing RTC time: 12/6/2017 21:26:9
GetIsRunning Before: 1
IsDateTimeValid Before: 1
GetIsRunning After: 1
IsDateTimeValid After: 1
GetIsRunning End: 1
IsDateTimeValid End: 1
-- NTP Server hostname: time.nist.gov
-- Transmit NTP Request
Sync frequency set low
Summer Time
Succeccful NTP sync at 21:26:19 12/06/2017
Syncing RTC time: 12/6/2017 21:26:19
GetIsRunning Before: 1
IsDateTimeValid Before: 0
GetIsRunning After: 0
IsDateTimeValid After: 0
GetIsRunning End: 0
IsDateTimeValid End: 0

ansarhun, 2017-11-20:
Hello,
I have the same problem, after NTP call, i try to update the RTC and after that IsDateTimeValid always returns false. I have to unplug from USB and remove battery so I can re-set it to a valid date.

Do you have any solution or workaround?

Thanks.

Makuna, 2017-11-21:
Can you provide a minimum sketch that reproduced the problem?  I will try to debug it.

One work around could be if GetIsRunning() returns false, then call begin() again and see if that resets it.

ansarhun, 2017-11-21:
Thank you for the suggestion will try it and replace the battery also.
If the issue still persists I will prepare and attach a minimal solution that reproduces the issue.

ansarhun, 2017-11-21:
Okay so the issue seems to be resolved:

No change after
 - battery replace 
 - calling begin again

After some googling I found the following issue:
https://github.com/esp8266/Arduino/issues/1025

So I always restarted the NodeMCU and after the restart the error occurred (NTP was called on boot as well). 

Applying the fix: http://www.forward.com.au/pfod/ArduinoProgramming/I2C_ClearBus/index.html
seems like solved the problem.

As I understand the I2C lines were left in a wrong state and couldn't recover from it after a reset.

Thanks for the help!

loadletter, 2018-05-24:
I've also ran across a similar issue while trying to switch between multiple I2C buses, tried a couple of the I2C recovery methods and a modified Wire library with no success.

It appears that doing Wire.begin() with a new set of pin also leaves the previous ones in a wrong state (using multiple Wire objects didn't help).

changing my program from
- RTC I2C begin/access
- secondary I2C begin/access
- RTC I2C begin/access
- deepsleep

to

- secondary I2C begin/access
- RTC I2C begin/access
- RTC I2C access
- deepsleep

Solved the issue, hope this can help others.

shaddow1201, 2018-05-24:
I've had the same issue with it running by itself, when i ran checking routines in loop without a delay.   Once i put second device on I2C it seems to have helped a lot.  I will check code but, i think i inadvertently came across that same soln as loadletter.

Makuna, 2018-08-26:
This seems to be esp8266 specific and may have been fixed with the latest Esp8266 Arduino release (2.4.2).
But I am leaving this issue open as it contains valuable information for finding a solution until others have reported it as no longer a problem.

nikuz, 2018-09-23:
Same issue on ESP32. It's hard to debug, cause you need to wait a few hours after device boot to get false on IsDateTimeValid. I'll write my solution if I'll solve this.

Makuna, 2018-09-23:
Something to try, can put another device on the bus.  Some of the 3231 have a eeprom on the module and thus bus, some don't.

nikuz, 2018-09-23:
Already tried three modules of two different types. One of this: https://ebay.to/2OLqUd0, and two of this: https://ebay.to/2I1UfNX. Same behaviour. They works perfectly if don't set time initially by NTP.

Makuna, 2018-09-23:
@nikuz How you set the time isn't really the issue, the problem is more than likely that WiFi is effecting the I2C communications (Wire) that are used to communicate with the RTC.  Have you brought this up on the Esp32-Arduino group?  I do see lots of issues with i2C, but nothing specific to this problem yet.

Makuna, 2019-02-24:
Esp32 seems to still have issues https://github.com/espressif/arduino-esp32/issues/1998

Makuna, 2020-07-25:
closing due to being stale (and related bugs in  esp32 have also either been fixed or gone statel).

NOTE:  The error handling has been improved since this issue was created and calling `LastError()` in this case should provide a hint as to why it would return false.

closed 2020-07-25 by Makuna