Makuna/Rtc #43: ESP32 - Undefine uint8, uint16 ... Error

yim1969, 2017-09-08:
I tried this module with ESP32 and got some error,

Undefine uint8, uint16 ... Error

found a problem with sequence of include file in RtcDateTime.CPP. 

#include "RtcDateTime.h"
#include <Arduino.h>

Then I change my local copy to 

#include <Arduino.h>
#include "RtcDateTime.h"

It works. 

Leave this info comment and advise that my resolution is correct or not. I am new to Git.

Makuna, 2017-09-08:
Its actually uint8_t correct?  not just uint8?

From a quick glance, I would say you were right; but let me check this on other platforms.

yim1969, 2017-09-08:
yes. please check. 

By the way, Thank you for your code !!!

Makuna, 2017-09-08:
I could not reproduce this.  Do you try the samples and see if they reproduce it?

Please check the library version, board support version, and your Arduino IDE version.  

p.s.  I am cleaning up some of the includes and esp32 stuff due to reviewing and finding it a little messy (especially the samples).

Robiv8, 2017-09-09:
Are you sure this was the problem?
because that
#include "RtcDateTime.h"
#include <Arduino.h>
is the same as
#include <Arduino.h>
#include "RtcDateTime.h"

except the sequence.

Makuna, 2017-09-09:
@Robiv8 I did not say this was the issue, I stated I could not reproduce it.  The changes I made were based on my style guide (standard includes before local includes).

Makuna, 2018-05-06:
I am closing this as I can not reproduce the issue and no contact has been made to state it is still a problem.

closed 2018-05-06 by Makuna