Makuna/Rtc #46: Creating interface for decoupling from T_WIRE_METHOD

klyman, 2017-09-08:
Use case that I am referring to is like this. I have a Task which is using RtcDS1307 class. In my particular case I am using TwoWire class as a low level implementation. When I am passing RtcDS1307 instance in constructor I have to write something like this:
```
class Task {
public:
  Task(RtcDS1307<TwoWire>& rtc) {}
}
```
It causes leak between abstraction layers. Since RtcDS1307 interface is independent from T_WIRE_METHOD I would like to have some parent class with virtual methods. Having such I would be able to use:
```
class Task {
public:
  Task(RtcAccessible& rtc) {}
}
```
which decouples from TwoWire class.
Simple implementation coule be like this:
```
class RtcAccessible {
public:
  virtual ~RtcAccessible() {}
  virtual bool IsDateTimeValid() = 0;
  ...
};

template<class T_WIRE_METHOD> class RtcDS1307: public RtcAccessible
{
public:
  bool IsDateTimeValid() override {
    return GetIsRunning();
  }
  ...
```

If you are willing to add such feature I can create a pull request for you.

Makuna, 2017-09-08:
interfaces and virtual functions are not a design I wish to incorporate into my library due to how memory wasteful they are.  Remember, this library is used on some of the smallest AVRs that have very limited memory.

most modern designs use the "contains a" concept over the older "is a"; and the use of template specialization is the method I use to accomplish this.

in the case you mention above, I would expect that task to be the owner/manager of the Rtc, so it would contain it and there is no "leak".

klyman, 2017-09-08:
I understand resource saving explanation and actually expected it.

To your suggestion - I have multiple tasks which uses the same Rtc object so it cannot contain it. Actually it would be much stronger coupling. Maybe I will go with template pattern for decoupling like you do.

Thank you for your answer. You can close the issue.

closed 2017-09-08 by Makuna