Makuna/Rtc #47: Arduino IDE

bobybobybob, 2017-10-11:
Hello,
the Arduino IDE say

there is a update for Makuna RTC Version 2.0.1

after Update it say there is still a update. Do i install 2.0.0 no update found but next start it say again 2.0.1 found and when i install this it go into cyrcle....

Makuna, 2017-10-12:
Hope did you install the first one?  Through GitHub or through the ide?

bobybobybob, 2017-10-12:
Sorry is too long time ago. I just see sometime the window that there are updates and that i can update to  2.0.0 but 2.0.1 is installed. I look to your homepage and see that 2.0.2 is the most actual Version. 

I have now removed it by hand from the arduino libary folder harddrive. Restart and want to install your libary again but very crazy, the Makuna RTC is still here and now it show it is 2.0.2 and no update. Look like there was two Versions at same time installed.

bobybobybob, 2017-10-12:
Problem solved. Thank you for great Makuna RTC Libary :-)

closed 2017-10-12 by bobybobybob