Makuna/Rtc #49: Class 'RtcTemperature' /  RtcDS3231::GetTemperature() Rework Needed

mrwgx3, 2017-12-11:
Hi Makuna

Enjoyed studying your RtcDS3231 library; I found your use of classes and templates most instructive. While studying the temperature processing code, however, I became most puzzled on how raw temperature was converted to a float. A side-by-side comparison of the D3231 datasheet and your code lead me to realize that the rather ambiguous description of the temperature register function admits (2) plausible interpretations:

```
Temperature Registers (11h–12h)
Temperature is represented as a 10-bit code with a resolution of 0.25°C and is accessible at
location 11h and 12h. The temperature is encoded in two’s complement format. The upper 8 bits,
the integer portion, are at location 11h and the lower 2 bits, the fractional portion, are
in the upper nibble at location 12h. For example, 00011001 01b = +25.25°C.
```
The 1st interpretation is that the upper and lower registers reflect a 16-bit wide 2's complement number, which reflects the temperature scaled by a factor of 256.

The 2nd case would be that the upper byte is an 8-bit 2's complement number representing a signed integer temperature of range -128 to 127, and the lower byte is an 8-bit unsigned integer holding a scaled fractional portion, 256 * { 0.00, 0.25, 0.50, 0.75 } = {0x00, 0x40, 0x80, 0xC0 }.

The bit diagrams accompanying the register description support the former, and the example within the description, the latter (too bad the example wasn't signed).

In order the resolve the issue, both interpretations were coded, and a dusty can of freeze spray was dug out of the closet. After cooling the RTC to some negative temperature, the correct interptation should show a monotonically increasing sequence of numbers as the part warms up. __The 1st interpretation, the scaled 16-bit 2's complement temperature, won the day__. The test code which generated the table below is included at the end of this post.

```
Column     Interpretion / Description                      Function
------------------------------------------------------------------------------
  A        original library code                           AsFloat()
  B        test code,  8-bit int/uint interpretation       AsFloatTestA()
  C        test code, 16-bit 2's complement word           AsFloatTestB()
  D        tweaked library code, added after initial test  AsFloat(), modified 

-12 degC  through -10 degC
--------------------------
   A        B           C        D
-12.00   -12.00      -12.00   -12.00    degC
-12.25   -12.25      -11.75   -11.75    degC
-12.50   -12.50      -11.50   -11.50    degC
-12.75   -12.75      -11.25   -11.25    degC
-11.00   -11.00      -11.00   -11.00    degC
-11.25   -11.25      -10.75   -10.75    degC
-11.50   -11.50      -10.50   -10.50    degC
-11.75   -11.75      -10.25   -10.25    degC
-10.00   -10.00      -10.00   -10.00    degC

Transition through 0 degC
-------------------------
  A       B          C        D
-1.00   -1.00      -1.00   -1.00    degC
-1.25   -1.25      -0.75   -0.75    degC
-1.50   -1.50      -0.50   -0.50    degC
-1.75   -1.75      -0.25   -0.25    degC

0.00   0.00      0.00   0.00    degC      Above 0 degC, all cases the same
0.25   0.25      0.25   0.25    degC
0.50   0.50      0.50   0.50    degC
0.75   0.75      0.75   0.75    degC
1.00   1.00      1.00   1.00    degC
```

Interestingly enough, the library code only requires a minor tweak to function correctly:
```
// Corrected 'AsFloat()' routine
   float AsFloatTestC ( RtcTemperature& rtmp )
   {
      float degrees = (float)rtmp.AsWholeDegrees();;
      degrees += (float)rtmp.GetFractional() / 100.0f ;
      return( degrees );
   
   } //AsFloatTestC

versus

   float AsFloat()
   {
[... 211 lines cut ...]
//
float AsFloatTestB ( RtcTemperature& rtmp )
{
// Retrieve MS portion of 2's comp. temperature
   int8_t   msbyte = rtmp.AsWholeDegrees();

// Reverse libary scaling to recover LS portion
   uint8_t lsbyte = ( (rtmp.GetFractional() / 25) << 6 );

   int16_t  itemp  = ( (uint8_t)msbyte << 8 ); // Scale upper byte, now 16-bit integer
            itemp += lsbyte;                   // Concant the lower byte
   return( (float)itemp / 256.0 );             // Un-scale and return
   
} //AsFloatTestB

// Corrected 'AsFloat()' routine
float AsFloatTestC ( RtcTemperature& rtmp )
{
   float degrees = (float)rtmp.AsWholeDegrees();;
   degrees += (float)rtmp.GetFractional() / 100.0f ;
   return( degrees );
   
} //AsFloatTestC

```

smz, 2017-12-31:
@mrwgx3, thank you for pinging me about this!

I've been out of country for several weeks and couldn't look into this matter until now.

I will surely incorporate your modified logic into my [Arduino-RTCtime](https://github.com/smz/Arduino-RTCtime) library ASAP.

I take the opportunity for wishing you, @Makuna and anybody else reading a very happy 2018.

Thanks!

mrwgx3, 2017-12-31:
Thank you, @smz, for the holiday wishes! As I'm rather new to pull-requests, I don't know what's automatically communicated when making/reviewing/working with them. Hence I'm letting you know that I did submit a pull-request with corrections to Makuna's library (will mention @Makuna just in case!), with the assumption you can easily use it. If this is not the case, I would be willing to do a pull request with corrections for your library.

Thanks, and have a Happy New Year
mrwgx3

mrwgx3, 2018-01-03:
Have closed original pull-request (#50) and opened up a new pull-request (#51). The old PR had too many changes to be managed easily (still learning the ropes)

Makuna, 2018-01-23:
Ok, thinking about the model to expose the temperature values, I keep getting back the core requirement is to not require floats; but the inherent problem of dealing with a negative value in the range of one degree is not solvable easily without them.  By introducing the separate concept of sign that you have done is an attempt but it makes it harder for comparisons and math (even though they were flawed before).

I think the core of your work is still valid, it is just the exposure of the integer values is the problem.  To solve this I am proposing a two step solution.

**First**, let us finalize your work and get it merged.  The only thing holding it up (beyond naming) is the method to expose the three parts, sign, integer, and fractional.  I propose this.
```
int8_t sign() // returns -1 or 1 for sign.  
uint8_t integer() // returns the unsigned whole  integer of the value, you must apply sign yourself
uint8_t fractional() // returns the fractional value (0, 25, 50, 75)
```
I would also change the sample code to not use printf and instead use direct value prints (printf is also heavy in code).

**Second**, I will come back and create another pull that adds operators to the RtcTemperate class for subtract, add, and comparisons so the user will directly use it rather than try to convert it before doing any of them.  Further, I will investigate adding a print to it that will minimize the complexity of printing the temperature.

mrwgx3, 2018-02-05:
@Makuna 
```
First, let us finalize your work and get it merged.
```
Have pulled your plate off the stack, will be back soon with commentary on your last post...

mrwgx3, 2018-02-07:
@makuna
```
Ok, thinking about the model to expose the temperature values, I keep getting back the core
requirement is to not require floats; but the inherent problem of dealing with a negative value
in the range of one degree is not solvable easily without them. By introducing the separate
concept of sign that you have done is an attempt but it makes it harder for comparisons and
math (even though they were flawed before).
```
The problem has been already solved by the RS3231 RTC itself, you don't need floats. The concatenation of DS3231 temperature registers R11/R12 is the __primary__ math object, __not the float derived from it__. It has a sign, magnitude, and you can do math (including comparisons) natively with the compiler. You need nothing else!

From the commentary on your proposed (2) step solution, I'm guessing you still want to use the following variable triad as your primary math object, considering your willingness to write math operators using it:
```
int8_t sign() // returns -1 or 1 for sign.  
uint8_t integer() // returns the unsigned whole  integer of the value, you must apply sign yourself
uint8_t fractional() // returns the fractional value (0, 25, 50, 75)
```
There is a simpler way to avoid the need for cumbersome scaling in order to do math, one hinted at by the 'fractional()' routine:

Just use a __decimal__ scaling (x100) for integer temperatures, as it is quite easy to multiply by 100 in one's head.

It's almost as easy to do the scaling in software, just right-shift (6) places to convert to the native temperature scaling from 256 to 4, then multiply by 25 to achieve a (x 100) scaling. I would let the constructor handle the 256-to-4 re-scaling, then change all the other routines to use the (x4) scale.

The integer temperature method looks like:
```
    // Integer temperature, scaled to decimal degrees
    int16_t AsDecimalDegC()
    {
       return( 25 * scaledDegC );  // Scale change from (x4) to (x100)
    }
```
It's easy to do human-friendly math and comparisons:
```
  degreesF = (int16_t)(  ( 320000 + td.AsDecimalDegC() * (uint32_t)180 ) / 100  );  // Fahrenheit conversion
  if ( degreesF > 21200 )
      Serial.print( "My tea is boiling..." );
```
__I would only use the sign/int/frac triad methods only for printing.__

```
I would also change the sample code to not use printf and instead use direct value
prints (printf is also heavy in code).
```
See attached sketch (text format), DS3231_TemperatureB.ino.txt, for example print methods.
```
I will investigate adding a print to it that will minimize the complexity of
printing the temperature.
```
All changes discussed here have been put into my fork of your library, in branch ['getTemp_rework_take5'][rtc_rework_final]. The attached sketch demonstrates the use of the new methods.

[DS3231_TemperatureB.ino.txt](https://github.com/Makuna/Rtc/files/1704716/DS3231_TemperatureB.ino.txt)

Review the code, test it, rename things however you wish.

[rtc_rework_final]: https://github.com/mrwgx3/Rtc/tree/getTemp_rework_take5  "Temperature Rework"

Makuna, 2018-02-08:
Then how about this then...
 
1) Leave the internal representation as you have reworked it.
2) Expose only two methods to get the temperature.  GetAsFloatDegreesC(), and GetAsCentiDegreesC().  The second is centridegrees C; using the metric prefix to define a unit that for the 100th degree value.   REMOVE all others.
3) Avoid printf.  The code is VERY large to implement it.  Targeting AVR 8bit chips it is best to avoid it in samples.

mrwgx3, 2018-02-08:
I'm a little baffled? Did you read the code?
```
Avoid printf. The code is VERY large to implement it.
Targeting AVR 8bit chips it is best to avoid it in samples.
```
Aside from the single 'snprintf_P' statement that you used in 'printDateTime(), there is not a single 'printf' statement in the attached test sketch. Only Serial.print() and Serial.println() statements.
```
Further, I will investigate adding a print to it that will minimize the complexity of printing
 the temperature.
```
In 'RtcTemperature.h', did you see [the centridegreesC  to ASCII string conversion routines][centri_ascii]? They are low-level 'C', and independent of data content for a given RtcTemperature instance. Do you wish to keep/discard the method and associated test routine? I'm assuming my 'investigation' convinced you to discard, given you last comments.
```cpp
// Print a floating point representation of temperature using Serial.print()
// and the default stringAsDecimalDegC() method
//
void printDecimalDegC ( RtcTemperature& td )
{
  Serial.print( td.stringAsDecimalDegC() );

} //printDecimalDegC
```
Do you still want to keep the [sign/int/frac triad methods][sif_triad]? (I would keep). If you do, what do you want to call them? How about its test code?
```cpp
// Print a floating point representation of temperature using Serial.print()
// and the GetSign(), GetUnsignedDeg(), and GetFractional() methods.

void printSignIntFrac (  RtcTemperature& td )
{
  uint8_t  ii = td.GetUnsignedDeg();  // Integer portion,    0 to 128
  uint8_t  ff = td.GetFractional();   // Fractional portion, 0, 25, 50, or 75

  // Selectively add spaces to maintain const. field width
  if ( ii  < 100 ) Serial.print( " " );
  if ( ii  <  10 ) Serial.print( " " );
  if ( ii >=   0 ) Serial.print( " " );

  // Print sign and integer portion
  Serial.print( (td.GetSign() < 0) ? '-' : ' ' );
  Serial.print( ii );

  // Print decimal and fractional portion
  Serial.print( "." );
  if ( ff == 0 )
    Serial.print( "00" );
  else
    Serial.print( ff );

} //printSignIntFrac
```
Do you still want the test example on how to print a float temperature?

```cpp
// Print a floating point temperature using Serial.print() and
// the AsFloatDegC() method
//
void printFloatDegC ( RtcTemperature& td )
{
  float  tf = td.AsFloatDegC();

  // Selectively add spaces to maintain const. field width
  if ( abs(tf) <  100 ) Serial.print( " " );
  if ( abs(tf) <   10 ) Serial.print( " " );
  if (     tf  >=   0 ) Serial.print( " " );

  Serial.print( tf, 2 );  // Prints a signed float

} //printFloatDegC
```
I need some feedback here, specific and to the point:
1. Do you even want a test sketch to show the usage of the RtcTemperature methods? If you don't, I'll close the pull request.
2. Tell me by name the RtcTemperature methods you wish to keep, and give them their proper names. I generate another branch with updates for merging into the original PR (you've given me twp)

P.S. Here is the output of the test sketch, which I used to verify print formatting.
[TemperatureB_Output.txt](https://github.com/Makuna/Rtc/files/1705654/TemperatureB_Output.txt)

[centri_ascii]: https://github.com/mrwgx3/Rtc/blob/getTemp_rework_take5/src/RtcTemperature.h#L73#L142

[sif_triad]: https://github.com/mrwgx3/Rtc/blob/getTemp_rework_take5/src/RtcTemperature.h#L52#L72

Makuna, 2018-02-08:
(sorry, but I am only getting small amounts of time to work on this, so I can only do small snipits at a time of reading, coming up to speed again, etc).  Its easier to view PRs than complete samples.

Pull way back, lets break this down into smaller PRs.  I wont submit a new library version until its complete.

First, just put in the code to switch to the new internal format and remove all but the float accessor.  This means also remove any others like decimal and fractional accessors.  Forget about the sample for now; that will get cleaned up last.

mrwgx3, 2018-02-08:
Yes, my time's limited too, hence wanting to wrap all this up. Per your directions, have issued a more atomic PR, #57

Makuna, 2018-02-16:
merged your work and my work in.  Still needs some testing and final temperature sample

Makuna, 2018-05-06:
release 2.1.0 addresses this.

closed 2018-05-06 by Makuna