Makuna/Rtc #55: Democode with Timezone Lib?

bobybobybob, 2018-01-20:
Do you have Domecode that use your RTC + Timezone Lib from Christianson?

I have try to combine them but arduino crash after  
    setSyncProvider(Rtc.GetDateTime()); 

(RTCDemo from timezone + yourds3231simple)

I ask Mr. Timezone but he only wrote nothing salvation.

If Timezone is incompatible, ok then itisinkompatible

Makuna, 2018-01-22:
This is a question, which is better asked on the gitter channel.

But to answer it, no I have not worked with the timezone lib.

In general, the best approach is to work in GMT time for everything except where you finally show it to your user; at which point you convert it right before display. 

I am not sure why it would be incomptible, its more of finding what format it requires and providing that.  The RtcDateTime object exposes the common EPOCH format through APIs (`Epoch32Time()` & `Epoch64Time()`) which I am sure they support in some fashion.

Makuna, 2023-06-07:
Epoch32Time is deprecated due to poor name.  It is replaced with two other sets..

Unix32Time() and InitWithUnix32Time()
Ntp32Time0 and InitWithNtp32Time()