Makuna/Rtc #56: DateTime persistent in RTC?

Nic2012, 2018-02-05:
Hi,
I have a tiny RTC with a DS3231(with Battery), using on RaspPi3 works fine, after restarting the Pi, RTC clock shows correct current datetime.
But its different when i use it with a NodeMCU v2 with the following code:

```
#include <Wire.h>       //I2C library
#include <RtcDS3231.h>  //RTC library

RtcDS3231<TwoWire> rtcObject(Wire);
volatile int setRtc = 0;
const int interruptPin = 0; //GPIO 0 (Flash Button)

void handleInterrupt() {
  detachInterrupt(interruptPin);
  setRtc = 1;
}

void setup() {
  Serial.begin(115200);
  rtcObject.Begin();
  pinMode(interruptPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin), handleInterrupt, FALLING);
}
 
void loop() {
  if (setRtc == 1) {
    rtcObject.SetDateTime(RtcDateTime(18, 02, 05, 18, 0, 0));
    setRtc = 0;
    Serial.println("rtc was set");
  }
  RtcDateTime currentTime = rtcObject.GetDateTime();
  char str[19];
  sprintf(str, "%04d-%02d-%02d %02d:%02d:%02d",
          currentTime.Year(), currentTime.Month(), currentTime.Day(),
          currentTime.Hour(), currentTime.Minute(), currentTime.Second()
         );
  Serial.println(str);
  delay(1000);
}
```

After a pause of around 20sec (unplug from power) when I restart the Esp8266 again, then the RTC clock starts with a delay of 20sec.
Maybe I have overseen sth. Whats is needed that the RTC  shows current dateitime also when the ESP is not powered in the meantime? Any help is appreciated.
Best regards,
Nic

Makuna, 2018-02-05:
Its hard to understand exactly what your issue is.  Do you have the debug output?

Are you stating that the time is delayed by 20 seconds? 
Are you stating that time never progresses?

BTW, in your setup code, you should wait for the serial to attach, like this...
```
Serial.begin(115200);
while (!Serial); // wait for serial device to attach
```

Nic2012, 2018-02-05:
Forget my issue, I think there was an unlucky constellation of testing, maybe the code in the interrupt handling was running without my knowledge when i repower the ESP again. Maybe it has set datetime always new. But after simplifying the code, it works like it should.
Sorry, and thanks for y fast response.
Best regards,

closed 2018-02-05 by Nic2012