Makuna/Rtc #61: dayOf Sunday

kleinmantara, 2018-04-29:
Hi,

thanks for the library - i have a question about the dayOf
```c
RtcDateTime dt = rtc.GetDateTime();
dt.DayOfWeek() // returns 0 on sunday and 6 on saturday
```

When i create a alarm with ```dayOf=0``` for sunday the alarm is **never** triggered. If i instead use ```dayOf=7``` for sunday the alarm is triggers.
```c
DS3231AlarmOne alarm1(
    dayOf,       //day 0-6 > Sunday=0
    alarmHour,   // hour 0-23
    alarmMinute, // minute 0-59
    0,           // second 0-59
    DS3231AlarmOneControl_HoursMinutesSecondsDayOfWeekMatch);
```
So for alarms the following values are correct - but the ```dt.DayOfWeek()``` methods returns the value in the comments.
```
const byte Monday    = 1; //1
const byte Tuesday   = 2; //2
const byte Wednesday = 3; //3
const byte Thursday  = 4; //4
const byte Friday    = 5; //5
const byte Saturday  = 6; //6
const byte Sunday    = 7; //0 <-- diffrent
```
Bug, feature - or my mistake?

Makuna, 2018-04-29:
Its a bug, or more accurately an inconsistency in my API.

The actual RTC hardware uses 1-7 for day of week.  But early DS1307 had bugs in calculating day of week after the year 2000 so I ignore the day of week value and calculate my own for the RtcDateTime and I designed around 0-6 inclusive as its more useful when working in C/C++.
The problem here is that the Alarm values are directly used/written into the RTC hardware.  So they use 1-7.

I will work on a fix.  I want to keep it 0-6

Makuna, 2018-04-29:
I just merged in fix.  If you want to try it, you will need to use github (see readme, its not too hard).  I still need to do more testing before making a new release with it which could take a week.

Makuna, 2018-05-06:
release 2.1.0 addresses this.  The Wiki was also updated.

closed 2018-05-06 by Makuna