Makuna/Rtc #63: Can not change date and time

SCherkass, 2018-04-30:
Maybe I'm spupid, but I can not change the time and date in the chip. The current date value is read normally (seconds change), but attempts to write to the microcircuit do not change date and time. Tell me please what I'm doing wrong.

Code: 
Init part:
#include <Wire.h>
#include <RtcDS3231.h>
RtcDS3231<TwoWire> Rtc(Wire);

setup(){
    Wire.begin(3,1);
    Rtc.Begin();
    Rtc.Enable32kHzPin(false);
    Rtc.SetSquareWavePin(DS3231SquareWavePin_ModeNone); 
    RtcDateTime setVal;
    setVal = RtcDateTime(18, 4, 30, 11, 54, 0);
    Rtc.SetDateTime(setVal);
}

loop(){
  RtcDateTime now = Rtc.GetDateTime();
  String  retvalstr = printDateTime(now);
....... and another code........
}

I see date and time 01.01.2000 08:07:55 and time changes, ie the chip works

Makuna, 2018-04-30:
Please note that the readme (at the front of the this repo, points you to use gitter for questions.

You set the year to 18, not 2018.  

use your printDateTime to print the datetime before you call setdatetime to see what it really is.

SCherkass, 2018-05-01:
Hello Michael,

Thank you for answer.  Of course I checked both of methods of initialisation (setVal = RtcDateTime(18, 4, 30, 11, 54, 0); and setVal = RtcDateTime(2018, 4, 30, 11, 54, 0);) with same results. And of corse i checked variable after initialisation. Initialisation of variable gone correctly. I checked it with printDateTime(). Problem only in  Rtc.SetDateTime(setVal); I checked a datasheet DS3231 for find write protect pin or write protect bit and not found nothing about it. That is reason because i write here...

Makuna, 2018-05-01:
Have you tried the samples and see if they work?

SCherkass, 2018-05-02:
Hello Michael,

Sample DS3231_Simple work properly! I will try to understand - where is problem in my code. Thank you for support!

closed 2018-05-02 by SCherkass