Makuna/Rtc #65: Rtc.SetDateTime(buf);  gives error in compiling phase with Arduino ide

axfatec, 2018-05-10:
https://github.com/Makuna/Rtc/blob/5b8849193205e6050b098a4a945482de47823b28/examples/DS3231_Simple/DS3231_Simple.ino#L112

Hello, I'm not so expert, with few months of trial.  
I'm tring to updated RTC 3231 with date get from UTC (ntp server) from here:
---
  unsigned long epochTime = timeClient.getEpochTime();
  utc = epochTime;
  TimeChangeRule CEST = { "CEST", Last, Sun, Mar, 2, 60 };     //Central European Summer Time
  TimeChangeRule CET = { "CET ", Last, Sun, Oct, 3, 0 };       //Central European Standard Time
  Timezone CE(CEST, CET);
  local = CE.toLocal(utc);
-----
then I convert it to format compatible to DS3231 RTC and try to store inside it:
-------
char buf[20]; 
snprintf_P(buf, countof(buf), PSTR("%02u/%02u/%04u %02u:%02u:%02u"), month(local),
day(local),year(local), hour(local), minute(local), second(local)); 
Serial.println(buf); 
Rtc.SetDateTime(buf);
----------
unfortunately I got an error compling it with Arduino ide

exit status 1
'countof' was not declared in this scope

I tried also this way:
..
snprintf(buf, sizeof(buf),....

but I got this error:

exit status 1
invalid user-defined conversion from 'char [20]' to 'const RtcDateTime&' [-fpermissive]

So, what I forget or I do wrong ?
THANKS
---------

axfatec, 2018-05-10:
I just find a solution with this call:
Rtc.SetDateTime(RtcDateTime(year(local),month(local),day(local), hour(local), minute(local), second(local)));

Makuna, 2018-05-10:
Several things:
* For questions, follow the Readme and use gitter; issues are for bugs not questions.
* Never set the RTC with local time; always use Zulu time.  Convert to local time right before you show the user.  This also goes for storing/logging time, always log Zulu time.  
* If you have a epoch time, you can call `RtcDateTime.InitWithEpoch32Time()` or `RtcDateTime.InitWithEpoch64Time()`.

closed 2018-08-25 by Makuna