Makuna/Rtc #68: DS3231 RTC with NodeMCU

sandymantvs, 2018-06-26:
I am using Nodemcu with ds3231.  While i use the Storeint program, the output in the serial monitor is given below.

> EªiYªP‚,jHRRY–¡VÕY¬µJW¥¡Õ¥Õš•YJªTE�uEZB\ÊH
!HYµ…I‹)VeWU ÖA…ðZ¬eA¬TYUeZ!\“
Š*jU!\K
EKŠZ¬Õ±WT_¥compiled: Jun 26 201809:57:54
06/26/2018 09:57:54
RTC lost confidence in the DateTime!
RTC was not actively running, starting now
85/165/2009 37:165:
 Rtc ready for storage <<<

This is the result from the serial monitor.  the date and time is set differently.  When i used the same module with arduino mega, it was perfect but with using Nodemcu it is not perfect.  But the ds3231 simple program is working fine (That is showing the date and time stored in ds3231).  But adjusting the time is not fine with nodemcu.

Makuna, 2018-06-26:
Please include relevant versions of esp 8266 Arduino and rtc library.
What is storeint?

SimoDolcio, 2018-08-25:
hi I have the same error of sandymantvs.
if I try to read the date I receive this 85/165/2009 37: 165:
I use NodeMCU (ESP-12e) with arduino IDE 1.8.5 an the library is 2.1.1.

thanks 
regards

SimoDolcio, 2018-08-25:
thanks for reply and for library....
1. Esp Arduino Board support versione = 2.4.1
2. RTC hardware: https://www.ebay.it/itm/Shield-DS-3231-modulo-RTC-real-time-clock-memoria-I2C-Arduino-PIC-ART-CL18/232002075742?hash=item360466bc5e:g:xcYAAOSwbYZXelri
`
void setup() {
  rtc.Begin();
  if (!rtc.GetIsRunning()) {
    Serial.println("RTC was not actively running, starting now");
    rtc.SetIsRunning(true);
  }
  // never assume the Rtc was last configured by you, so
  // just clear them to your needed state
  rtc.Enable32kHzPin(false);
  rtc.SetSquareWavePin(DS3231SquareWavePin_ModeNone);
  RtcDateTime ntpDateTime = RtcDateTime(588493899);
    rtc.SetDateTime(ntpDateTime);
    delay(1000);
    printDateTime(rtc.GetDateTime());
}
#define countof(a) (sizeof(a) / sizeof(a[0]))
void printDateTime(const RtcDateTime& dt) {
    char datestring[20];
    snprintf_P(datestring, 
            countof(datestring),
            PSTR("%02u/%02u/%04u %02u:%02u:%02u"),
            dt.Month(),
            dt.Day(),
            dt.Year(),
            dt.Hour(),
            dt.Minute(),
            dt.Second() );
    Serial.println(datestring);
}
`

SimoDolcio, 2018-08-27:
thanks for your support.
I updated esp board and your library and now all works fine

regards