Makuna/Rtc #80: Outputs gibberish

jians5, 2019-02-12:
Here's the output after uploading: ⸮⸮A⸮C⸮⸮⸮⸮⸮⸮1S⸮_⸮RA6⸮W#⸮⸮a⸮⸮8⸮٣⸮⸮<⸮⸮⸮É⸮⸮

It continues adding more and more random characters.

Makuna, 2019-02-14:
Please follow the guidelines for creating "issues".  This is where active bugs in the library are created; not a place for asking questions.  In the Wiki and the Readme both point you toward the Gitter channel for asking questions.

What were you uploading?  What device?  
This looks like the baud rate listed in the sketch doesn't match the baud rate in the serial monitor.

Closing due to being just a question.

closed 2019-02-14 by Makuna

jians5, 2019-02-14:
Sorry - noob here. Noted for the future. Thanks for your work on the
project, and the baud rate might just be it! Not sure how you would guess
that but If I understood the genius mind I wouldn't be asking stupid
questions in the wrong place on Github.