Makuna/Rtc #81: RtcDS3234.h - line 307 _spi.write => _spi.transfer (?)

fmudruino, 2019-02-17:
**Describe the bug**
Compiling examples for DS3234 with arduino 1.8.8 fails with reference to missing "write" ('class SPIClass' has no member named 'write')

**Development environment (please complete the following information):**
 - OS: [Win10]
 - Build Environment [Arduino IDE v.1.8..8]
 - Board target [Nano]
 - Library version [v2.3.0] 

**Minimal Sketch that reproduced the problem:**
examples that come with library

**Additional context**
I changed line 307 to _spi.transfer(...) and it seems to work. (alarm example compiles und runs as expected)

fmudruino, 2019-02-17:
not sure if misclicked (I didn't close on purpose) - so I reopen - just for letting you know and fix it in the source - I assume with the simple code change proposed it will be fixed...

Makuna, 2019-02-17:
Esp has the write method, its not available on many other platforms.  It should have been transfer.
https://github.com/Makuna/Rtc/pull/82

Makuna, 2019-02-17:
fixed in 2.3.1 release (which may take a day to make it through the release process)

fmudruino, 2019-02-18:
thanks for quick fix/update.

closed 2019-02-18 by fmudruino