Makuna/Rtc #88: Impossible values in RtcDateTime. Wrong month

Martin-Laclaustra, 2019-03-31:
**Describe the bug**
The library accepts all 0s in:
`RtcDateTime(uint16_t year, uint8_t month, uint8_t dayOfMonth, uint8_t hour, uint8_t minute, uint8_t second)`
https://github.com/Makuna/Rtc/blob/b2252020a3b4d93b6b20a4cbede0e20329b8d6f4/src/RtcDateTime.h#L30
... which is interpreted as the following date and time:
`00/00/2000 00:00:00`
with clear wrong day-of-month 0 of month 0.
The problem goes further as:
https://github.com/Makuna/Rtc/blob/b2252020a3b4d93b6b20a4cbede0e20329b8d6f4/src/RtcDateTime.cpp#L101
returns: `1367256704` (which is greater than current century seconds)
Where does this number come from?
https://github.com/Makuna/Rtc/blob/b2252020a3b4d93b6b20a4cbede0e20329b8d6f4/src/RtcDateTime.cpp#L103
with arguments uint16_t 0,0, and 0 overflows when `-1` (as int16_t) appears in:
https://github.com/Makuna/Rtc/blob/b2252020a3b4d93b6b20a4cbede0e20329b8d6f4/src/RtcDateTime.cpp#L87
and then it returns the value (uint16_t) `65535`.
When it is further processed in:
https://github.com/Makuna/Rtc/blob/b2252020a3b4d93b6b20a4cbede0e20329b8d6f4/src/RtcDateTime.cpp#L104
The value is casted to uint32_t and then multiplied in:
https://github.com/Makuna/Rtc/blob/b2252020a3b4d93b6b20a4cbede0e20329b8d6f4/src/RtcDateTime.cpp#L92
which results in a value of `65535×24×60×60=5662224000`, which overflows, again, the uint32_t that this function returns `5662224000 & 0xFFFFFFFF = 1367256704`.

**To Reproduce**
This happened with the example DS1302_Simple.ino with the un-initialized DS1302 chip. When new (or after removing the back-up battery) this chip actually provides impossible day and month (0) values.
The script never updates the correct time because the comparison in:
https://github.com/Makuna/Rtc/blob/b2252020a3b4d93b6b20a4cbede0e20329b8d6f4/examples/DS1302_Simple/DS1302_Simple.ino#L42
is never true.

**Expected behavior**
Although the main script could program what to do in this case, it seems natural that the library would be able to deal with and recover from these impossible values (which appear in the real-world scenario of removing-replacing the back-up battery).
I am unsure which approach would be best. Some options:
- Probably the best: raise/return some error. Not easy, however, because the values are provided in the constructor. Afterwards, let the script deal with the error. Although it should be actively used, maybe just a member function that checks whether day or month are 0 would be enough, or (why not) a member function that checks that all values are in their sensible range. Something like:
`uint8_t Valid() const {return _dayOfMonth!=0 && month!=0;}`
or
```
uint8_t Valid() const {return month>=1 && month<=12 &&
_dayOfMonth>=1 && _dayOfMonth<=31 && // this could be more detailed
_hour<=23 && _minute<=59 && _second<=59;}

```
- I do not know how the RTC adder will increment the month after 24h, or when the month will change. Assuming that both will increase to 1 after 24h and 31 days, then it would be sensible to interpret the date as 30th November 1999. After 24h it would be day 1 of month 0 of 2000 (1st December 1999). Not easy either, as the library can not hold dates before the year 2000, and also probably not worth working in this direction, as this date is never going to be used.
- May be the most feasible thing. Also the most transparent, but certainly obscure behaviour, would be to automatically "correct" wrong values of 0 to 1 for day and month. Writing them back to the RTC. There could be a bool setting in the RTC constructor called "autocorrect", to always allow more conservative reads.

**Development environment (please complete the following information):**
 - OS: Linux
 - Build Environment: Arduino IDE v.1.8.3
 - Board target: ESP8266
 - Library version: v2.3.2

**Minimal Sketch that reproduced the problem:**
I checked the overflow problems with a c++ program:
```
#include <iostream>
const uint8_t c_daysInMonth[] = { 31,28,31,30,31,30,31,31,30,31,30,31 };
template <typename T> T DaysSinceFirstOfYear2000(uint16_t year, uint8_t month, uint8_t dayOfMonth)
{
    T days = dayOfMonth;
    for (uint8_t indexMonth = 1; indexMonth < month; ++indexMonth)
    {
        days += c_daysInMonth[indexMonth - 1];
    }
    if (month > 2 && year % 4 == 0)
    {
        days++;
    }
    return days + 365 * year + (year + 3) / 4 - 1;
}
template <typename T> T SecondsIn(T days, uint8_t hours, uint8_t minutes, uint8_t seconds)
{
    return ((days * 24L + hours) * 60 + minutes) * 60 + seconds;
}
uint32_t TotalSeconds(uint8_t _yearFrom2000, uint8_t _month, uint8_t _dayOfMonth,uint8_t _hour, uint8_t _minute, uint8_t _second)
{
    uint16_t days = DaysSinceFirstOfYear2000<uint16_t>(_yearFrom2000, _month, _dayOfMonth);
    return SecondsIn<uint32_t>(days, _hour, _minute, _second);
}
int main() {
  std::cout << TotalSeconds(0,0,0,0,0,0);
}
```

**Additional context**
Thank you for all your work.
Probably related to the last line of the OP in #60.

Makuna, 2019-03-31:
The 1302 doesn't provide a status flag for invalid date/time like the others.

The normal model to follow is to call `IsDateTimeValid()` before trusting what it returns; but again 1302 doesn't expose this.  But you are stating that in this case it will return all zeros?  It is not documented to do that.  But it might be able to be used to implement a a IsDateTimeValid method.

Martin-Laclaustra, 2019-03-31:
This sketch explores the initial status:
```
#include <ThreeWire.h>  
#include <RtcDS1302.h>

ThreeWire myWire(D4,D5,D2); // IO, SCLK, CE

const uint8_t DS1302RawRtcSize = 8;

uint8_t GetRawRtc(uint8_t* pValue, uint8_t countBytes)
{
    uint8_t countRead = 0;

    myWire.beginTransmission(DS1302_REG_TIMEDATE_BURST | THREEWIRE_READFLAG);

    while (countBytes > 0 && countRead < DS1302RawRtcSize)
    {
        *pValue++ = myWire.read();
        countRead++;
        countBytes--;
    }

    myWire.endTransmission();

    return countRead;
}

void setup () 
{
    Serial.begin(115200);
    Serial.println();
}

void loop () 
{
    // read data
    uint8_t buff[DS1302RawRtcSize];
    const uint8_t count = sizeof(buff);
    uint8_t gotten = GetRawRtc(buff, count);

    Serial.print("data read (");
    Serial.print(gotten);
    Serial.print(") = ");
    for (uint8_t ch = 0; ch < gotten; ch++)
    {
        Serial.print("0x");
        Serial.print(buff[ch],HEX);
        Serial.print(" ");
    }
    Serial.println();
    
    delay(5000);
}
```
This prints out the following:
```
data read (8) = 0x80 0x0 0x0 0x0 0x0 0x0 0x0 0x0 
data read (8) = 0x80 0x0 0x0 0x0 0x0 0x0 0x0 0x0 
data read (8) = 0x80 0x0 0x0 0x0 0x0 0x0 0x0 0x0 
...
```
If the stop flag is cleared with `Rtc.SetIsRunning(true);`, it starts counting from there:
```
data read (8) = 0x5 0x0 0x0 0x0 0x0 0x0 0x0 0x0 
data read (8) = 0x10 0x0 0x0 0x0 0x0 0x0 0x0 0x0 
data read (8) = 0x15 0x0 0x0 0x0 0x0 0x0 0x0 0x0 
...
```
Maybe this could be added to "RtcDS1302.h"
```
uint8_t IsDateTimeValid() const {
RtcDateTime vtest = GetDateTime();
return vtest.Month()>=1 && vtest.Month()<=12 &&
vtest.Day()>=1 && vtest.Day()<=31 && // this could be more detailed depending on month
vtest.Hour()<=23 && vtest.Minute()<=59 && vtest.Second()<=59;}
```
... and this to the example sketchs:
```
if (!Rtc.IsDateTimeValid()) 
{
    // Common Cuases:
    //    1) first time you ran and the device wasn't running yet
    //    2) the battery on the device is low or even missing

    Serial.println("RTC lost confidence in the DateTime!");
    // following line sets the RTC to the date & time this sketch was compiled
    // it will also reset the valid flag internally unless the Rtc device is
    // having an issue

    Rtc.SetDateTime(compiled);
}
```

gb0101010101, 2019-04-03:
Great minds think alike @Martin-Laclaustra. Just ran into this problem myself on a project that uses this library and though I would check this project to see if anyone else noticed it. Your bug report is the best I have *ever* seen. Wish I had people reporting bugs to me with this level of depth!

I figured it would be better to have RtcDateTime return an invalid date response before trying to set and validated the invalid date/time on RtcDS1302. I ended up adding a validation function RtcDateTime.isValid() just like you suggested. It checks everything including day is valid for month and leap year for Feb 29th. See my commit (other project) for new function and example of how I used it.
https://github.com/gb0101010101/SDL_ESP8266_WeatherPlus/commit/fb02d3907f50735e7a0d981877f11607362a59cc#diff-240a911f425670089ed85d278fae5b31

Makuna, 2019-04-11:
https://github.com/Makuna/Rtc/pull/90

Martin-Laclaustra, 2019-04-12:
Thank you!
I was about to ask you if you would need a pull request...
I was thinking in refining it a little more. In:
https://github.com/Makuna/Rtc/blob/c47da87daa01cadff3731cbee8e09f5f28f07481/src/RtcDateTime.h#L55
`(_dayOfMonth > 0 && _dayOfMonth <= pgm_read_byte(c_daysInMonth + indexMonth - 1)) + ((year & 3) == 0 && ((year % 25) != 0 || (year & 15) == 0)) &&
`
... but you would need to move the function to the .cpp file.
This would complete a total check of the validity.
I hope you like it and consider using it.
Source for the leap year added day [code](https://stackoverflow.com/questions/3220163/how-to-find-leap-year-programatically-in-c/11595914#11595914).

Makuna, 2019-04-12:
Have you tried setting the time with an invalid leap year date with your RTC hardware.  My understanding is that it will convert it to the valid "progressed time" date time and will never return an invalid leap year date.

Martin-Laclaustra, 2019-04-13:
TLDR; After some tests (below), it is worth having a full validity check in the class.
The clock accepts invalid values in all fields and even rolls them over.
I created a script to stress test each field. I set to print the date in ISO 8601 format. This is only an excerpt:
```
void loop () 
{
    RtcDateTime newtime;
    RtcDateTime now;
    
    newtime = RtcDateTime(0,0,0,0,0,0); //year,month,dayOfMonth,hour,minute,second
    Rtc.SetDateTime(newtime);
    for(int i=5; i>0; i--){
      now = Rtc.GetDateTime();
      printDateTime(now);
      delay(2000);
    }
    Serial.println();

    newtime = RtcDateTime(1,1,1,1,1,66); //year,month,dayOfMonth,hour,minute,second
    Rtc.SetDateTime(newtime);
    for(int i=5; i>0; i--){
      now = Rtc.GetDateTime();
   ...

}
```
Tested combinations:
```
(0,0,0,0,0,0)
(1,1,1,1,1,66)
(1,1,1,1,62,56)
(1,1,1,25,59,56)
(1,1,32,23,59,56)
(1,13,28,23,59,56)
(1,13,30,23,59,56)
(1,13,31,23,59,56)
(1,2,29,23,59,56)
```
Results:
```
2000-00-00 00:00:00
2000-00-00 00:00:02
2000-00-00 00:00:04
2000-00-00 00:00:06
2000-00-00 00:00:08

2001-01-01 01:01:66
2001-01-01 01:01:68
2001-01-01 01:01:70
2001-01-01 01:01:72
2001-01-01 01:01:74

2001-01-01 01:62:56
2001-01-01 01:62:58
2001-01-01 01:63:00
2001-01-01 01:63:02
2001-01-01 01:63:04

2001-01-01 25:59:56
2001-01-01 25:59:58
2001-01-01 26:00:00
2001-01-01 26:00:02
2001-01-01 26:00:04

2001-01-32 23:59:56
2001-01-32 23:59:58
2001-01-33 00:00:00
2001-01-33 00:00:02
2001-01-33 00:00:04

2001-13-28 23:59:56
2001-13-28 23:59:58
2001-13-29 00:00:00
2001-13-29 00:00:02
2001-13-29 00:00:04

2001-13-30 23:59:56
2001-13-30 23:59:58
2002-04-01 00:00:00
2002-04-01 00:00:02
2002-04-01 00:00:04

2001-13-31 23:59:56
2001-13-31 23:59:58
2002-04-01 00:00:00
2002-04-01 00:00:02
2002-04-01 00:00:04

2001-02-29 23:59:56
2001-02-29 23:59:58
2001-02-30 00:00:00
2001-02-30 00:00:02
2001-02-30 00:00:04
```

Martin-Laclaustra, 2019-04-19:
The need for a good validity check in RtcDateTime is not only for checking the time from RTC but also for checking that an external time is valid before setting it in the RTC.

Makuna, 2019-04-21:
See latest code (not released yet), it now tests leap year

Martin-Laclaustra, 2019-04-21:
Thank you for the attention you paid to this problem.
The code seems fine.

You wrote a code efficient when discarding wrong values `(_dayOfMonth < 32)`, `(_dayOfMonth > 29)`, which should be the infrequent case, but that adds additional tests to the most frequent case, that is when the time value is valid.
I would focus in a straightforward check of the validity for all variables, only branching into detailed chech if _dayOfMonth > 28.
This should be efficient because of [short-circuit evaluation](https://en.wikipedia.org/wiki/Short-circuit_evaluation). Note that `Year()` will be only called twice for leap years and three times for the last years of each century.
Anyway this is only academic efficiency, as this code would not be called so often.
If you allow me, I would propose this version, also in order to achieve a clearer logic.

```
bool RtcDateTime::IsValid() const
{
    // this just tests the most basic validity of the value ranges
    // and valid leap years
    // It does not check any time zone or daylight savings time
    return !(
        (_month < 1 || _month > 12) ||  // invalid month
        (_dayOfMonth < 1 ||             // invalid _dayOfMonth
            (_dayOfMonth > 28 &&           // efficiency ward
                _dayOfMonth > (            // compute detailed limit
                    _month != 2 ? (           // not February
                        ((((_month - 1) % 7) % 2) == 0) ?
                        31 :
                        30
                    ) : (                     // February
                        ((Year() & 3) == 0 && ((Year() % 25) != 0 || (Year() & 15) == 0)) ?
                        29 :                  // leap year
                        28                    // not leap year
                    )
                )
            )
        ) ||
        (_hour > 23) ||                 // invalid hour
        (_minute > 59) ||               // invalid minute
        (_second > 59)                  // invalid second
        )
}
```

Makuna, 2019-04-21:
You have point and I appreciate the feedback, but I do not find your example of code very readable so I can't just take it as is.

Martin-Laclaustra, 2019-04-21:
I apologize. It was an effort.
Here is another shot:
```
bool RtcDateTime::IsValid() const
{
    // this just tests the most basic validity of the value ranges
    // and valid leap years
    // It does not check any time zone or daylight savings time
    return !(                  // false if any invalid value
        (_hour > 23)           // invalid hour
        ||
        (_minute > 59)         // invalid minute
        ||
        (_second > 59)         // invalid second
        ||
        (_month < 1)           // invalid lower month limit
        ||
        (_month > 12)          // invalid upper month limit
        ||
        (_dayOfMonth < 1)      // invalid lower dayOfMonth limit
        ||
        (                      // invalid upper dayOfMonth limit
            (_dayOfMonth > 28) // if false, valid for any month
            &&                 // do not check the next in that case,
            (_dayOfMonth >     // else, compute detailed upper limit
                (_month != 2 ?
                    // not February: formula == 0 for months with 31 d.
                    ( ((((_month - 1) % 7) % 2) == 0) ? 31 : 30 )
                    :
                    // February: formula == true for leap years.
                    ( ((Year() & 3) == 0 && 
                        ((Year() % 25) != 0 || (Year() & 15) == 0)) ? 29 : 28 )
                )
            )
        )
    )
}
```

gb0101010101, 2019-04-23:
Sorry have not tested proposed code but when using RTC3231 before I noticed that any year < 2000 is altered to 2000 plus last two digits. e.g. 1969 becomes 2069.  Would this invalidate leap year calculation that are not based off 2000 an use Year()?

Martin-Laclaustra, 2019-04-23:
The code should be valid for years counted in our era (4 digits).
The class can not hold any year before 2000, because the clocks can not hold them either. See the class definition:
https://github.com/Makuna/Rtc/blob/f287c1ee11f436c26b6a72d856877ecdddb0003b/src/RtcDateTime.h#L26
I understand that this class is written as a support class for interacting with the clocks, not as a general means for holding dates in programs.

Makuna, 2019-04-23:
"I understand that this class is written as a support class for interacting with the clocks, not as a general means for holding dates in programs."  

A correct and accurate statement.  If the RTC ds1302 had an invalid flag like all the other chips, the method would still not be present on the RtcDateTime as it is out of scope of what this library is for.

Martin-Laclaustra, 2019-04-24:
In addition to checking whether the time provided by the RTC is valid...
Would it not be also useful to make sure that you are about to set valid times to the other RTC models? How do other clocks behave facing tests like the ones I performed [above](https://github.com/Makuna/Rtc/issues/88#issuecomment-482792152)?
Of course, using it before setting the clock would be completely optional.
Thank you Makuna for your attention and all you work in the lib.
Your work inspires many of us.

closed 2019-05-29 by Makuna

Martin-Laclaustra, 2019-05-29:
I assume that you did not like the code in https://github.com/Makuna/Rtc/issues/88#issuecomment-485285665 either.
Thanks for writing a fix and I hope that next time I can contribute code deemed more useful.

Makuna, 2019-05-29:
I thought I included it?  Examine it again, if you don't see something similar present, go ahead and reopen this.

Martin-Laclaustra, 2019-05-31:
All the functions are in the code. It was just my last attempt to clarify the validation routine (Apr 22 comment) that we did not talk any more about. There are probably more important tasks around.
Again, thanks, and we can have a look at that code in some moment if there is time (or need) to do it.