Makuna/Rtc #98: Crash in Arduino 1.8.9 with alarm example (ISR not in IRAM!) on ESP8266

pilnikov, 2019-07-31:
Hello.
When i try using alarm example (with IRQ), my esp8266 is crashed with message "ISR not in IRAM!"
I was able to resolve this by adding
` ICACHE_RAM_ATTR` in
`void InteruptServiceRoutine()` declarations
To become so
`void ICACHE_RAM_ATTR InteruptServiceRoutine()`

Makuna, 2019-07-31:
This is a very specific nuance to programming Esp8266 and Esp32 with no general solution within the context of Arduino to manage it.

How did you come to find the solution?

pilnikov, 2019-07-31:
https://github.com/esp8266/Arduino/issues/6127

pilnikov, 2019-07-31:
The problem appeared along with the new (1.8.9) version of the Arduino. In 1.8.5 everything was fine

Makuna, 2019-08-01:
https://groups.google.com/a/arduino.cc/forum/#!topic/developers/56VPC9t2MPc

pilnikov, 2019-08-01:
Maybee add a comment about .... ?

Makuna, 2019-11-07:
https://github.com/Makuna/Rtc/pull/102
I just added my own cross platform method attribute, and updated the examples

Makuna, 2020-02-22:
v2.3.4

closed 2020-02-22 by Makuna