Makuna/Rtc #100: disturbance

mehrdad987, 2019-10-23:
hi
I don't know it's my code problem or its a bug
I am using 2 interrupts and 7 task Is this the reason for the disturbance?

`void taskNine( void * parameter )
{

  RtcDateTime now = Rtc.GetDateTime();
  printDateTime(now);

  RtcTemperature temp = Rtc.GetTemperature();
  //float Device_Temperature = 
  //temp.Print(Serial);
  // you may also get the temperature as a float and print it
  Device_Temperature = (temp.AsFloatDegC());
  Serial.println(Device_Temperature);
  
    Serial.println("------------------------------------------------------------------------------making file-Ending task 7");
  vTaskDelete( NULL );

}`

`10:15:10.862 -> 10/22/2019 17:53:34
10:15:10.862 -> 
10:15:10.862 -> Message appended
10:15:10.862 -> 0
10:15:10.862 -> ---making file-Ending task 7
10:15:10.862 ->---Ending task 1
10:15:11.850 -> Frequency   inf
10:15:11.850 ->   RPM  4294967266
10:15:11.850 -> 0
10:15:11.850 -> ---Ending task 5
10:15:11.850 -> 85/165/2009 37:165:
10:15:11.850 -> ⸮
10:15:11.850 -> Message appended
10:15:11.850 -> 23
10:15:11.850 -> ---making file-Ending task 7---Ending task 1
10:15:12.000 -> 
10:15:12.000 -> Temperature = 10
10:15:12.000 -> ---Ending task 2
10:15:12.875 -> 10/45/2019 30:78:78
10:15:12.875 -> -
10:15:12.875 -> 23
10:15:12.875 -> ---making file-Ending task 7
10:15:12.875 -> Message appended
`

mehrdad987, 2019-10-23:
debuger

10:57:58.525 -> [I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffd2db8
10:57:58.525 -> [I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
10:57:58.525 -> [I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
10:57:58.559 -> [I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
10:57:58.559 -> [I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=5
10:57:58.559 -> [I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffd2ecc bits=112
10:57:58.559 -> [I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffd30f4
10:57:58.559 -> [I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffd2ea8
10:57:58.559 -> [I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
10:57:58.559 -> [I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
10:57:58.593 -> [I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
10:57:58.593 -> [I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
10:57:58.593 -> Temperature = [I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
10:57:58.593 -> [I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
10:57:58.593 -> [I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
10:57:58.593 -> [E][WiFiClienti2c.c:1130] i2cProcQueue(): Bus busy, r[D][HTTPClient.cpp:1003] connect():  connected to devices.hiots.net:443
10:57:58.626 -> [I][esp32-hal-i2c.c:1130] i2cProcQueue(): Bus busy, reinit
10:57:58.626 -> [W][esp32-hal-i2c.c:1411] i2cCheckLineState(): invalid state sda(21)=0, scl(22)=1
10:57:58.626 -> [D][esp32-hal-i2c.c:1419] i2cCheckLineState(): Recovered after 1 Cycles

Makuna, 2019-10-23:
Please provide more details.  You didn't provide ANY of the basic information you should be.  What platform, what sketch, what versions, and you didn't even describe the problem you are seeing.

closed 2019-11-07 by Makuna

mehrdad987, 2019-11-09:
RTC lost confidence in the DateTime..and it's not a battery
it might be a problem with I2C and SPI 
or because of the ISR
OR WIFI
I'm working on it

mehrdad987, 2022-11-22:
> Please provide more details. You didn't provide ANY of the basic information you should be. What platform, what sketch, what versions, and you didn't even describe the problem you are seeing.

Hello, has anyone seen this disturbance?---> 
11:11:33.885 -> 22-11-2022 11:11:30
11:11:34.875 ->165-85-2009 37:165:
11:11:35.911 ->45-11-2022 30:78:78
11:11:36.895 ->45-11-2022 30:78:79

mehrdad987, 2022-12-13:
it happen beacuse low size of ---char datestring[20];

`#define countof(a) (sizeof(a) / sizeof(a[0]))
void printDateTime(const RtcDateTime& dt)
{
  /* =========== main TD ============= */
  char datestring[30];
  snprintf_P(datestring, countof(datestring), PSTR("%02u-%02u-%04u %02u:%02u:%02u"),
             dt.Day(),
             dt.Month(),
             dt.Year(),
             dt.Hour(),
             dt.Minute(),
             dt.Second() );

  timeSec = (datestring);
}`