Makuna/Rtc #109: DS3231 - interrupt on alarm is not cleared

Tost69, 2020-02-23:
Hi,

I use your lib to set alarms on a DS3231 clock module and it works well so far.
But there is an issue with the clearing of the interrupt:

- use Rtc.SetAlarmOne(alarm1); to set the alarm, the interupt pin is HIGH
- after some time the alarm is deployed and the interrupt pin is set to LOW
-> result is that if an external device is wake up by the interrupt, the fake wake-ups occures

I use an ESP32, set it to sleep, the alarm wakes the ESP and any further try to set the ESP to sleep again will fail because of the not cleared interrupt.

Is there something which has to be written to the DS3231 to reset the interupt?

Thanks and kind regards, Tom

Makuna, 2020-02-24:
First, when you created this issue, did you read the options.  This is a question, and those should be directed to the gitter channel.

Once you handle the alarm, you need to call the `LatchAlarmsTriggeredFlags()`.  If you look at the example for the DS3231 alarms you will see how this is done.

Tost69, 2020-02-24:
First: thanks for clarification and second: yes you are right, I missed to use LatchAlarmsTriggeredFlags() after the ESP wakeup (I already used LatchAlarmsTriggeredFlags() within the alarms handler but I forgot to call it in the setup after wakeup). So thanks for your quick help and kind regards, Tom

closed 2020-02-24 by Tost69