Makuna/Rtc #110: ESP8266, No DS3231 connected at all, GetIsRunning() and IsDateTimeValid() both return True.

OfferLa, 2020-04-09:
**NOTE:  If you are seeking help or have questions, this is NOT the place to do it.**  For questions and support, jump on Gitter and ask away.  
[image]

**Describe the bug**
I've just tested now a code on an NodeMCU board.
At last it supposed to be connected to a DS3231 module, but now it's not connected to anything.
Nevertheless - println(Rtc.IsDateTimeValid()) and println(Rtc.GetIsRunning()) prints "1".

**To Reproduce**
Steps to reproduce the behavior:
I think I mentioned it: If you run the Arduino without the module exist - it will return true for both functions mentioned above.

**Expected behavior**
I think when there is no Rtc module present - both functions should return "false",
Because clock is not running, and time is not reliable.

**Development environment (please complete the following information):**
 - OS: Windows 10
 - Build Environment: Arduino IDE v.1.8.12
 - Board target [NodeMCU ESP-12 ] - you will need to check the board manager for this.
 - Library version [v2.3.3] - you will need to check the library manager for this.  If you used github to clone it, mention this and include the reference.

**Minimal Sketch that reproduced the problem:**
```
Serial.println("");
Serial.println("Starting RtcInitByCompile");

RtcDateTime compiled = RtcDateTime(__DATE__, __TIME__);

Serial.print("Compiled at: ");
Serial.println(compiled);

Serial.print("Rtc.IsDateTimeValid: ");
Serial.println(Rtc.IsDateTimeValid());

Serial.print("Rtc.GetIsRunning: ");
Serial.println(Rtc.GetIsRunning());

````

**Additional context**
Add any other context about the problem here.

Makuna, 2020-05-25:
https://github.com/Makuna/Rtc/pull/113
Please give this a try.

Makuna, 2020-06-24:
Please provide feedback if the fixes solve the issue for you.

OfferLa, 2020-06-29:
Really sorry for my delay.
Will test it (try to re-produce my original bug) again, and update soon.

wangmings, 2020-07-05:
Same question! It seems that this library is not well tested!

Question:

Error: 'printDateTime' was not declared in this scope
PrintDateTime (compiled);
^
Error: 'printDateTime' was not declared in this scope
PrintDateTime (now);

Makuna, 2020-07-05:
@wangmings This is not a forum to discuss unrelated issues.  This is a bug tracking mechanism.  Your post is not related to this issue.  If you want to ask questions jump on the gitter channel (see the readme/wiki for the link).  Further, the library does not even contain those methods.

Makuna, 2020-09-29:
new release includes the change

closed 2020-09-29 by Makuna