Makuna/Rtc #114: Check the status of the alarm

DVB-Arduino, 2020-06-14:
In the Jarzebski RTC library there is a function to check the status of the alarm - it is charged or worked. I did not find this in your library. If possible, please add.

Makuna, 2020-06-14:
Please include more information when you create issues.  What RTC module?

DS3231AlarmFlag LatchAlarmsTriggeredFlags()

This will reset the current alarm flags and return the status of the alarms.  Is this what you mean?

DVB-Arduino, 2020-06-14:
Sorry. DS3231 module. 
My device must be turned off for transportation. But the watch runs on battery power autonomously. When I turn on the device, I do not know if the alarm worked while the device was turned off. So I need a function that tells me if the alarm worked or not. 
But I do not need to reset the set alarm, as in DS3231AlarmFlag LatchAlarmsTriggeredFlags ().

Makuna, 2020-06-14:
What is your use case that this won't work for you?

The latching means the alarm can trigger again.   Just checking means it happened once in the past but if it did happen you won't know if it happens again after you check.  Further, the interrupt features won't trigger again unless you reset/latch them.

DVB-Arduino, 2020-06-14:
Thank you very much. This does not work as I thought. I'll think about how to set an alarm when you turn on the device. 
And yes, your library is cool!

Makuna, 2020-06-24:
Not exactly clear on your response, but it seems the issue is resolved.

closed 2020-06-24 by Makuna