Makuna/Rtc #118: Unbuildable RtcDS1302

Flavian, 2020-08-03:
**Describe the bug**
I downloaded the RTC lib with PlatformIO, and I tried to use it with ESP8266 board project. After the build steps the follow error message shown: 

".pio/libdeps/esp12e/RTC/src/RtcDS1302.h:173:61: error: 'THREEWIRE_READFLAG' was not declared in this scope
         _wire.beginTransmission(DS1302_REG_TIMEDATE_BURST | THREEWIRE_READFLAG);"

**Expected behavior**
A clear and concise description of what you expected to happen.

**Development environment (please complete the following information):**
 - OS: Ubuntu
 - Build Environment: PlatformIO the newest
 - Board target ESP8266
 - Library version: RTC@2.3.5

Thank you for your help.

Best reguards, Flavian!

Makuna, 2020-08-03:
You didn't include any of your code.  What are you including?
Did you try to compile one of the samples for the 1302?
```
#include <ThreeWire.h>  
#include <RtcDS1302.h>

ThreeWire myWire(4,5,2); // IO, SCLK, CE
RtcDS1302<ThreeWire> Rtc(myWire);
```

Flavian, 2020-08-03:
Thank you for your fast answare.

I have double checked my code and the error also appeared.

This is my code: 
`#pragma once

#include "TimerBase.h"
#include "Arduino.h"
#include <ThreeWire.h>  
#include <RtcDS1302.h>

class MyTime: TimerBase
{
private:
    RtcDS1302<ThreeWire> rtc;
public:
    MyTime(/* args */);
    unsigned long GetMillis() override;
    RtcDateTime GetTime() override;
    void SetTime(RtcDateTime time) override;
    ~MyTime();
};`

And this is the error messages: 

In file included from lib/DateTime/TimerBase.h:4:0,
                 from lib/DateTime/MyTime.h:3,
                 from lib/DateTime/MyTime.cpp:1:
.pio/libdeps/esp12e/RTC/src/RtcDS1302.h: In member function 'RtcDateTime RtcDS1302<T_WIRE_METHOD>::GetDateTime()':
.pio/libdeps/esp12e/RTC/src/RtcDS1302.h:173:61: error: 'THREEWIRE_READFLAG' was not declared in this scope
         _wire.beginTransmission(DS1302_REG_TIMEDATE_BURST | THREEWIRE_READFLAG);
                                                             ^
.pio/libdeps/esp12e/RTC/src/RtcDS1302.h: In member function 'uint8_t RtcDS1302<T_WIRE_METHOD>::GetMemory(uint8_t*, uint8_t)':
.pio/libdeps/esp12e/RTC/src/RtcDS1302.h:238:56: error: 'THREEWIRE_READFLAG' was not declared in this scope
         _wire.beginTransmission(DS1302_REG_RAM_BURST | THREEWIRE_READFLAG);
                                                        ^
.pio/libdeps/esp12e/RTC/src/RtcDS1302.h: In member function 'uint8_t RtcDS1302<T_WIRE_METHOD>::getReg(uint8_t)':
.pio/libdeps/esp12e/RTC/src/RtcDS1302.h:258:46: error: 'THREEWIRE_READFLAG' was not declared in this scope
         _wire.beginTransmission(regAddress | THREEWIRE_READFLAG);

Exactly I think this is the ESP8266 incompatible because I found some other error after I build ThreeWire.h. I think the PlatformIO does not insert the Arduino.h into the ThreeWire.h and this errors are appiered. These are just a little part of the errors. Could you help me about this?

**I have built the simpel code yet. I work on it**

"                                          
.pio/libdeps/esp12e/RTC/src/ThreeWire.h:47:32: error: 'delayMicroseconds' was not declared in this scope
             delayMicroseconds(1);     // tDC = 200ns
                                ^
.pio/libdeps/esp12e/RTC/src/ThreeWire.h:50:26: error: '_clkPin' was not declared in this scope
             digitalWrite(_clkPin, HIGH);
                          ^
.pio/libdeps/esp12e/RTC/src/ThreeWire.h:50:35: error: 'HIGH' was not declared in this scope
             digitalWrite(_clkPin, HIGH);
                                   ^
.pio/libdeps/esp12e/RTC/src/ThreeWire.h:56:33: error: 'INPUT' was not declared in this scope
                 pinMode(_ioPin, INPUT);
                                 ^
.pio/libdeps/esp12e/RTC/src/ThreeWire.h:56:38: error: 'pinMode' was not declared in this scope
                 pinMode(_ioPin, INPUT);
                                      ^
.pio/libdeps/esp12e/RTC/src/ThreeWire.h:59:35: error: 'LOW' was not declared in this scope
             digitalWrite(_clkPin, LOW);
                                   ^
.pio/libdeps/esp12e/RTC/src/ThreeWire.h: In member function 'void ThreeWire::resetPins()':
.pio/libdeps/esp12e/RTC/src/ThreeWire.h:95:17: error: '_clkPin' was not declared in this scope
         pinMode(_clkPin, INPUT);"

Flavian, 2020-08-03:
Some hot news...

I insert this line into the ThreeWire.h:"#include "Arduino.h"

I think the library for some reason can't handle the ESP8266. What is your opinion?

Makuna, 2020-08-03:
The library is an ARDUINO library.   While it should be compatible with the Platform IO build system, it is up to you the sketch author to understand how Platform IO is different.

The RTC1302 is not i2c, you would need to use another RTC chip.

Flavian, 2020-08-03:
The platformIO's compatible platform list contains the ESP8266, but you said this is only Arduino lib. So this is a little missleading for me. Thank you for your fast help. 

[This is the link for RTC on PlatformIO libs](https://platformio.org/lib/show/274/RTC?utm_source=platformio&utm_medium=piohome)

Makuna, 2020-08-03:
esp8266 is not the issue, it works fine with it.  It is basically a Arduino library that is compatible with Platform IO, if you follow standard Arduino compatibility rules, like include Arduino.h.

Also note, from your included sketch, you are not correctly defining the rtc object, see my included code above.

closed 2020-08-27 by Makuna