Makuna/Rtc #119: Time on Ds3231 gets wrong after some time.

Bredahl, 2020-08-27:
I run this piece of code where I set the time on a ds3231 module to the client time and then read the ds3231 module time to check that the time is set. It runs flawlessly with an older version of the AsyncTCP library, but when I run it under the latest version of the AsyncTCP library, the time on the Ds3231 module goes completely wrong after a short time.
Does anyone have an idea why this is happening.
[ds3231.txt](https://github.com/Makuna/Rtc/files/5134977/ds3231.txt)

Makuna, 2020-08-27:
This area is not for asking questions is for tracking bugs that need to be fixed.   It is not a forum. 

I am closing this as it is a question and the templates for bug report were not followed. 

I suggest that you jump on gitter and ask questions there.

closed 2020-08-27 by Makuna

Bredahl, 2020-08-27:
Ok, I'm really sorry for taking your time. I'm new to this. I will then just continue to use the old AsynTCP library.

Makuna, 2020-08-27:
Its the wrong way to do it.  jump on Gitter (message support system), the channel is listed in readme and wiki.  You will get more than just me on it to provide help.

Bredahl, 2020-08-29:
I have solved my problem with the AsyncTCP library with the code in the attachment.
[ds3231 Fixed.txt](https://github.com/Makuna/Rtc/files/5145109/ds3231.Fixed.txt)