Makuna/Rtc #120: DS3234 + ESP12F_Relay_X1 board:  "RTC lost confidence in the DateTime!"

T0m92, 2020-09-23:
Hi, i'm using your library for read a DS3234 date and time ( https://www.sparkfun.com/products/10160 )  from ESP12F_Relay_X1 
board ( http://www.chinalctech.com/m/view.php?aid=472 ).
I flashed the "DS3234_Simple" example sketch, the pinout i used is the following (DS3234 to ESP12F_Relay_X1):
GND to GND
VCC to 3V3 (also tried 5V pin)
SQW to GND
CLK to SCLK
MISO to MISO
MOSI to MOSI
SS to IO16 (DS3234_CS_PIN)
I'm using a new a fully charged battery for DS3234 module.
Battery model BR1225 3V from datasheet is requested CR1225 3V but i'm assumed it's the same thing.
When i open the serial monitor i get the following otuput:

09/23/2020 10:11:38
RTC lost confidence in the DateTime!
RTC was not actively running, starting now
RTC is older than compile time!  (Updating DateTime)
RTC lost confidence in the DateTime!
85/165/2009 37:165:
-0.25-0.25C
RTC lost confidence in the DateTime!
85/165/2009 37:165:
-0.25-0.25C
RTC lost confidence in the DateTime!
85/165/2009 37:165:
-0.25-0.25C
RTC lost confidence in the DateTime!
85/165/2009 37:165:
-0.25-0.25C
How i can solve this? Thanks in advance.

Makuna, 2020-09-23:
Please use Gitter channel (see readme/wiki) for questions.  The issues area is for Bug tracking of this source code NOT for asking questions or seeking help.

Did you search the issues for any past occurrences of a similar issues?  How about the Wiki?

More than likely you have not connected it correctly.  (See below).  Make sure the pins you set are exposed on the board correctly.  Note the CS0 pin on the board you have maybe the Esp8266 hardware controls the CS pin (unusual for Arduino) for the first device?

https://learn.sparkfun.com/tutorials/deadon-rtc-breakout-hookup-guide?_ga=2.246415236.1298702137.1600850979-973734605.1600850979

Unfortunately SPI has no communications error status that can be checked, so a scope or logic analyzer is the most useful to use to help diagnose.

T0m92, 2020-09-23:
Thanks for the quick response, I didn't know Gitter and I wrote here, sorry.
Anyway I read the links you passed me and you were right, it was a connection problem on wrong pins, now everything works correctly. Thanks again.

Makuna, 2020-09-23:
Added a FAQ for this issue here...
https://github.com/Makuna/Rtc/wiki/FAQ-%235

closed 2020-09-23 by Makuna