Makuna/Rtc #124: Method InitWithIso8601(const char* date) does NOT scan ISO8601 but rather RFC1123

brewmanz, 2021-05-28:
**Describe the bug**
Your commented example is "Sat, 06 Dec 2009 12:34:56 GMT", which is RFC1123. 
ISO8601 (which has many formats) would be more like "2009-12-06T12:34:56" or "2009-12-06T12:34:56+00:00" or "2009-12-06T12:34:56Z". 

**To Reproduce**
I just saw the code. 

**Expected behavior**
Either parse according to some ISO8601 format, or be renamed to InitWithRfc1123. 

**Development environment (please complete the following information):**
All. 

**Minimal Sketch that reproduced the problem:**
Use eyeball MkI. 

**Additional context**
None.

Makuna, 2021-05-30:
@techi602 provided this support.

Makuna, 2023-06-07:
`InitWithIso8601` is now deprecated method.  It has been replaced with `InitWithDateTimeFormatString` which allows the sketch author to provide a formatting string that defines how to covert the string.

Makuna, 2023-06-07:
v2.4.0

closed 2023-06-07 by Makuna