Makuna/Rtc #125: DS1302 TreeWire - This is not a issue, it is a question of "how?"

pilnikov, 2021-07-01:
I am trying to do this
`static RtcDS1302<ThreeWire> * ds1302;`
`static ThreeWire *  myTWire;`
`#define dio 9`
`#define clk 10`
`#define cs  11`
``
`void setup`
`{`
` myTWire = new ThreeWire(dio, clk, cs); // IO, SCLK, CE`
` ds1302 = new RtcDS1302<ThreeWire> (myTWire);`
`}`

And I get the following ...............

In file included from Z:\ardu\sketch\Test\test_ds1302\test_ds1302.ino:1:0:
Z:\ardu\sketch\libraries\Rtc_by_Makuna\src/RtcDS1302.h: In member function 'RtcDateTime RtcDS1302<T_WIRE_METHOD>::GetDateTime()':
Z:\ardu\sketch\libraries\Rtc_by_Makuna\src/RtcDS1302.h:173:61: error: 'THREEWIRE_READFLAG' was not declared in this scope
         _wire.beginTransmission(DS1302_REG_TIMEDATE_BURST | THREEWIRE_READFLAG);
                                                             ^~~~~~~~~~~~~~~~~~
Z:\ardu\sketch\libraries\Rtc_by_Makuna\src/RtcDS1302.h: In member function 'uint8_t RtcDS1302<T_WIRE_METHOD>::GetMemory(uint8_t*, uint8_t)':
Z:\ardu\sketch\libraries\Rtc_by_Makuna\src/RtcDS1302.h:238:56: error: 'THREEWIRE_READFLAG' was not declared in this scope
         _wire.beginTransmission(DS1302_REG_RAM_BURST | THREEWIRE_READFLAG);
                                                        ^~~~~~~~~~~~~~~~~~
Z:\ardu\sketch\libraries\Rtc_by_Makuna\src/RtcDS1302.h: In member function 'uint8_t RtcDS1302<T_WIRE_METHOD>::getReg(uint8_t)':
Z:\ardu\sketch\libraries\Rtc_by_Makuna\src/RtcDS1302.h:258:46: error: 'THREEWIRE_READFLAG' was not declared in this scope
         _wire.beginTransmission(regAddress | THREEWIRE_READFLAG);
                                              ^~~~~~~~~~~~~~~~~~
Z:\ardu\sketch\Test\test_ds1302\test_ds1302.ino: At global scope:
test_ds1302:4:18: error: 'ThreeWire' was not declared in this scope
 static RtcDS1302<ThreeWire> * ds1302;
                  ^~~~~~~~~
test_ds1302:4:27: error: template argument 1 is invalid
 static RtcDS1302<ThreeWire> * ds1302;
                           ^
test_ds1302:5:8: error: 'ThreeWire' does not name a type
 static ThreeWire * myTWire;
        ^~~~~~~~~
Z:\ardu\sketch\Test\test_ds1302\test_ds1302.ino: In function 'void setup()':
test_ds1302:9:1: error: 'myTWire' was not declared in this scope
 myTWire = new ThreeWire(9, 10, 11); // IO, SCLK, CE
 ^~~~~~~
test_ds1302:9:15: error: expected type-specifier before 'ThreeWire'
 myTWire = new ThreeWire(9, 10, 11); // IO, SCLK, CE
               ^~~~~~~~~
test_ds1302:10:24: error: 'ThreeWire' was not declared in this scope
 ds1302 = new RtcDS1302<ThreeWire> (myTWire);
                        ^~~~~~~~~
test_ds1302:10:33: error: template argument 1 is invalid
 ds1302 = new RtcDS1302<ThreeWire> (myTWire);
                                 ^
Using library Rtc_by_Makuna version 2.3.5 from: Z:\ardu\sketch\libraries\Rtc_by_Makuna 
exit status 1
'ThreeWire' was not declared in this scope

//------------------------------------------------------------------------------------------
How to properly initialize the Threewire object using the "new" operator ??

Makuna, 2021-07-01:
@pilnikov Did you read the text when you entered this issue?  This is not the place to ask questions, the Discussions is the place (along with Gitter or Discord) are the places to go to ask questions.  Links to all these locations are in the Readme and the top of the Wiki.

closed 2021-07-01 by Makuna

pilnikov, 2021-07-01:
ok