Makuna/Rtc #136: _yearFrom2000 contains 'year from 1970' with DS3231

de-arl, 2022-03-29:
Version 2.3.5 with ESP8266

The Year() method of the RtcDateTime object returns the current year + 30 with DS3231. This is caused by the _yearFrom2000 member variable containing the year from 1970 (=content of year variable from initializer argument). 

I tried to solve the problem but the program behaviour is weird. A quick fix to overcome the symptoms of the problem is to change the value of c_OriginYear in RtcDateTime.h line 22 from 2000 to 1970 
`
const uint16_t c_OriginYear = 1970;
`
and changing the initialization of _yearFrom2000 in the two-argument constructor of the RTCDateTime object iin RTCDateTime.cpp line 85 to add 30. Otherwise changing the value of c_OriginYear returns a wrong year in this kind of object.
`
RtcDateTime::RtcDateTime(const char* date, const char* time)
{
    // sample input: date = "Dec 06 2009", time = "12:34:56"
    _yearFrom2000 = StringToUint8(date + 9) + 30;
`

This fix is not a solution to the problem: the variable _yearFrom2000 still contains the year from 1970 and might lead to other problems as well, I don't know.

Trying to get the _yearFrom2000 variable right in RtcDs3231.h line 335 by adding 1970 instead of 2000 results in strange behaviour and does not work.
`
        uint16_t year = BcdToUint8(_wire.read()) + **1970**;
`

Please take a look at this.

Makuna, 2022-03-29:
The problem is more than likely how you are initializing something.  You didn't show any sample code so I don't know how you are using it.  
First, did you try the samples and did they work for you?
Second, its common that people mix up the constructor `RtcDateTime(uint32_t secondsFrom2000 = 0)` as EPOCH time, it is not.  They are about 30 years different.  If you need to init a RtcDateTime with EPOCH time based values, then use `InitWithEpoch32Time()` or `InitWithEpoch64Time()` methods. If you need to get an EPOCH time based values, the use `Epoch32Time()` or `Epoch64Time()`

de-arl, 2022-04-01:
Hi Makuna

Thanks for taking a look at this. I also took another look at the code. Looks as there was really a problem with the initialization. I did call the SetDateTime() method of the rtc object with normal epoch time `Rtc.SetDateTime(time(nullptr))`. As this function takes an RtcDateTime object as argument such a call calls the constructor you mentioned and sets the clock + 30 years to the future. 

I now set the clock like this:
```
    RtcDateTime settime;
    settime.InitWithEpochXXTime(time(nullptr));
    Rtc.SetDateTime(settime);
````
This works. 

I think I also found the issue for the strange behaviour of the program which occurs when changing some of the hard-coded constants.

The constructor of the RtcDateTime object assigns `uint16_t year` to `uint8_t _yearFrom2000`. This leads to undefined behaviour in variuos scenarios. As such scenarios are rare but happen, an application using this library might be unstable in the long term. 

Peace.

closed 2022-04-01 by de-arl