Makuna/Rtc #180: Missing declaration of countof() in RtcDateTime.h

TomRottler, 2023-07-01:
**Describe the bug**
I Added #include "RtcUtility.h" in RtcDateTime.h to get compiled the code in case of a compiler error for countof() .

I compiles your library in VS Code / PlatformIO

Please confirm.

TOM

TomRottler, 2023-07-01:
Hi,
I found the comment of cyberman after writing my issue and he claimed the same but solved it by setup the RtcUtility.h external but I guess it would be better to have it in the RTcDateTime.h because the call of countof() is there. Is this correct?

TOM

By the way: Great work!

Makuna, 2023-07-02:
The idea is that you don't include RtcDateTime.h directly, you include one of the RTCs (like RtcDS1302.h).  That header will include everything needed in order.  So RtcDateTime.h doesn't need to include it.

What case are you running into?

TomRottler, 2023-07-02:
Hi,

thank you for the response. It is a fork of the project from cyberman54 https://github.com/cyberman54/ESP32-Paxcounter.
He is using the Rtc library in some cases, even if no RtcDS3231 is installed.
I guess his project could be optimized at this point. I will contact him.

Best regards,

TOM

closed 2023-07-02 by TomRottler