Makuna/Rtc #184: time complied wrong

ririfonfon, 2023-07-05:
**Describe the bug**
when I upload my code. the monitor writes:
`compiled :       -> Monday 10/07/2023 10:17:57
now :            -> Monday 10/07/2023 10:17:59
Pure now :              742299479
Pure compiled :         742299477`

the system time is Wednesday 05/07/2023 10:17:57

**Development environment (please complete the following information):**
 - OS: edition Windows 10 Professionnel version 22H2 build 19045.3086
 - Processeur Intel(R) Core(TM) i9-7920X CPU @ 2.90GHz   2.90 GHz
 - Build Environment Visual Studio Code 1.79.2 PlatformIO Core 6.1.7 Home 3.4.4
 - Board target `[env:pico32]
platform = espressif32
board = pico32
framework = arduino
board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
board_build.flash_mode = qio
monitor_speed = 115200
board_build.partitions = partitions.csv
lib_deps = 
	links2004/WebSockets @ ^2.3.1
    fastled/FastLED @ ^3.6.0
	makuna/RTC @ ^2.3.5`.

Makuna, 2023-07-05:
Is this using the official released version or the GitHub master version?
i believe this was fixed yesterday but has not been in the official release.  If you could pick up the github master version it should be fixed.

ririfonfon, 2023-07-05:
Ok it's work.
with
`[env:pico32]
platform = espressif32
board = pico32
framework = arduino
board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
board_build.flash_mode = qio
monitor_speed = 115200
board_build.partitions = partitions.csv
lib_deps = 
	links2004/WebSockets @ ^2.3.1
    fastled/FastLED @ ^3.6.0
	https://github.com/Makuna/Rtc`

Makuna, 2023-07-06:
https://github.com/Makuna/Rtc/releases/tag/2.4.1

closed 2023-07-06 by Makuna