Makuna/Rtc #197: Expected ',' or ';' before 'enum'; Error compiling for board Arduino Uno

BusyBird15, 2023-09-04:
**Describe the bug**
I get the following error in the latest version:
```
In file included from /home/username/Arduino/libraries/Rtc_by_Makuna/src/RtcDS1302.h:30:0,
                 from /home/username/Desktop/HomeOS/HomeOS.ino:91:
/home/tyler/Arduino/libraries/Rtc_by_Makuna/src/RtcUtility.h:68:1: error: expected ',' or ';' before 'enum'
 enum Rtc_Wire_Error
 ^~~~
exit status 1
Error compiling for board Arduino Uno.
```
I also looked at the exact file in the exact place, and it did not reference enum; and everywhere "enum" was used, it was used correctly, no "," or ";" needed to be added.

**Development environment (please complete the following information):**
 - OS: Raspberry Pi OS 64-bit
 - Build Environment: Arduino IDE v.1.8.19
 - Board target: Built-in default AVR Board library, compiling on Arduino Uno Rev 3
 - Library version: 2.4.2

**Minimal Sketch that reproduced the problem:**
```cpp
#include <RtcDS1302.h>
ThreeWire myWire(io, clk, ce);
RtcDS1302<ThreeWire> Rtc(myWire);

void setup(){
  Rtc.Begin();
  RtcDateTime compiled = RtcDateTime(__DATE__, __TIME__);
}

void loop () {
  RtcDateTime now = Rtc.GetDateTime();
  Serial.println();
  Serial.print(now.Hour());
  Serial.print(":");
  Serial.print(now.Minute());
  Serial.print(":");
  Serial.print(now.Second());
  delay(1000);
}
```

Makuna, 2023-09-04:
First your minimal sketch doesn't compile due to the missing io,clk, and ce variables/constants.
Replacing those with some random pin numbers and it compiles fine.

What ever is before the include "RtcUtility.h" doesn't have a ; after a class or enum definition.  In RtcDS1302.h you can see that Arduino.h is included before it, so something in Arduino.h is causing the issue. 

But again, I can't repro this.

BusyBird15, 2023-09-05:
Thank you for your prompt response.

I had set the variables in the code (I forgot to include them in the sample), and that is not the issue.

Oddly enough, I changed the file `RtcUtility.h` from (at line 68):
```cpp
enum Rtc_Wire_Error
```
to
```cpp
;enum Rtc_Wire_Error
```
and the sketch compiled successfully. I'm not sure why, although I am not a C++ expert; so unless this causes me any upload errors once I finish the code, I suppose this fix is sufficient.

Makuna, 2023-09-05:
Sure, that's a workaround, but it's not the right solution.

This is an improper fix.  C++ you do not put semicolons in front of statements, they are at the end of the statement.  This is why I stated the problem lies in Arduino.h and not my header.  Something before didn't have the statement ending semi colon that it should, and it will affect other sketchs/libraries.

Due to me not be able to repro, it must be a RPI environment issue.

BusyBird15, 2023-09-05:
I tried compiling the code on my other Ubuntu (Linux) computer and got the same compiler error. I'll try with my fix, and I'll let you know if it uploads and functions properly or not.

BusyBird15, 2023-09-05:
Okay, upload worked fine and I have no issues with the function.

So, I guess I'll just leave the modified code as is.

closed 2023-09-05 by BusyBird15