Makuna/Rtc #209: warning: comparison of integer expressions of different signedness

kinlougha, 2024-05-13:
**Describe the bug**
Receive this warning when compile OnStepX under VSCode:

In file included from .pio\libdeps\teensy41\RTC\src\RtcDateTime.cpp:29:
.pio\libdeps\teensy41\RTC\src\RtcDateTime.h: In member function 'RtcDateTime RtcDateTime::operator+(int32_t) const':
.pio\libdeps\teensy41\RTC\src\RtcDateTime.h:168:41: warning: comparison of integer expressions of different signedness: 'int32_t' {aka 'long int'} and 'uint32_t' {aka 'long unsigned int'} [-Wsign-compare]        
  168 |         if (seconds < 0 && abs(seconds) > totalSeconds)

**Development environment 

 - OS: Window11
 - Build Environment VSCode v1.89.0
 - Board target Teensy 4.1
 - Library version 2.4.2

Correction:  Change "int32_t" for "uint32_t" on line 164 of RtcDateTime.h 

Thanks.

kinlougha, 2024-05-13:
OK I understand, i open a case to PJRC with this info and close this case.

Thanks.

kinlougha, 2024-05-14:
Hello,

This is the answer from PJRC : https://forum.pjrc.com/index.php?threads/warning-message-at-compile.75017/

Makuna, 2024-05-14:
https://github.com/Makuna/Rtc/pull/210

kinlougha, 2024-05-14:
Hi Michael,

I test the solution, with the master branch, on 2 platforms (Teensy 4.1 and ESP32) and no more warning from the library.

This case is close for me.

Thanks.

Makuna, 2024-05-14:
Leave it open (its marked as Pending).  I will close it when I publish the next update.

kinlougha, 2024-05-15:
Sorry Michael, I leave it open, I understand.

Thanks.

Makuna, 2024-11-04:
v2.5.0

closed 2024-11-04 by Makuna