Makuna/Rtc #221: compatibility with DS1338

TowerDefender007, 2024-12-16:
I would like if your library would also feature a compatibility for the DS1338 because its almost similar with the DS1307 but I cant tell if its exactly the same.

Makuna, 2024-12-16:
The only difference I see is that the DS1338 has a flag in bit 5 of the control register.  Even the device address is the same. It seems to have some duplicate features of the CH in the status register (shared with seconds register) but still requires to be set to zero to function.

[image]

You can use the DS1307 but just make sure to set the square wave which will also clear this flag, like ...
```
rtc.SetSquareWavePin(DS1307SquareWaveOut_1Hz); // or what ever you want the output pin to work
```

This should function until the library is updated specifically to support it.

TowerDefender007, 2024-12-17:
Okay thank you for your fast answer. I am looking forward for your update and will use your suggested way till then.